dusty-nv/jetson-inference #268: Would you mind if I port it to Windows?

vinjn, 2018-09-21:
I love this project, but I only work on desktop :)
And 90% time on Windows.

S4WRXTTCS, 2018-09-21:
Out of curiosity how would you go about porting it? I don't see a whole lot on using TensorRT for windows. Or maybe I'm not looking at the right place. 

Something like this would be great for Windows as it's an easy starting off point. In fact we had an intern who originally started trying to do inference on Windows, and I dragged him off it as his time was limited. I showed him this repository and he was able to accomplish what he wanted within a couple weeks. This included training the network, retraining, inference and some customization. 

It does look like it's getting easier to do inference in Windows using an NVidia GPU to accelerate the task. As an example Microsoft recently released there Machine Learning framework for Windows 10.  Where it supports acceleration on the CPU, GPU, or VPU (like the Movidius product). 

Even where it's at today I don't see as easy of a path as NVidia Digits for training, and the acceleration with this project. I'm not a Windows developer though. I'm hardware engineer that only deals with Software until I have a proof of concept, and then I hand it over. It's really funny trying to hand over a Linux only project to a bunch of C# Windows guys.

dusty-nv, 2018-09-21:
Hi @vinjn , @S4WRXTTCS   I do not believe TensorRT is available on Windows yet, which is a dependency of `jetson-inference`.

It looks like you can install DIGITS on Windows [here](https://github.com/NVIDIA/DIGITS/blob/master/docs/BuildDigitsWindows.md), with the caveat that the Caffe branch of Windows does not support DetectNet.

S4WRXTTCS, 2018-09-21:
@dusty-nv

In the tensorrt-5 survey there is a "which of the new features in TensorRT 5.0 are you looking for" question with one of the answers being Windows Support.  But, none of the downloads are for Windows. 

That's a little weird as everything else is in there. Hopefully it's just the RC that it's not in, and it will be in final one. Or someone really needs to change the answers to that question. :-) 

As to training I got to a point where I decided I was only going to use Containers. It's too much of a pain settings thing up, and then having conflicts. There is no support for GPU acceleration within Windows for containers. So I'm sticking with Ubuntu for Training.

vinjn, 2018-09-22:
Yes, TRT5 supports Windows.

vinjn, 2018-10-04:
TRT Windows package is released
https://developer.nvidia.com/nvidia-tensorrt-5x-download

closed 2023-04-27 by dusty-nv