dusty-nv/jetson-inference #280: C++ TensorRT Batch Inference gives wrong result

Optimus1072, 2018-11-28:
I am trying to run inference on multiple images using TensorRT API. Pseudo Code snippet for my application is

context.enqueue(batchSize, buffers, stream, nullptr);

Here,

buffers[0] = batchSize * INPUT_C * INPUT_H * INPUT_W
        buffers[1] = batchSize * outputSize

If I run with batchSize=1 , I get correct output but with batchSize > 1 , detection for images other than first image are wrong. Further, with batchSize=1, inference time is 7ms and for batchSize=3 it is around 16ms. So solving this issue will give a lot of boost to my application and as in general.

Can someone please suggest me what all things I can try to solve this issue. I am allocating memory to "buffers" this way

for (int b = 0; b < engine.getNbBindings(); b++)
    {
        DimsCHW dims = static_cast<DimsCHW&&>(engine.getBindingDimensions(b));
        size_t size = batchSize * dims.c() * dims.h() * dims.w() * sizeof(float);
        std::cout << "size of buff = " << size << std::endl;
        CudaCHECK(cudaMalloc(&buffers[b], size));
    }

dusty-nv, 2018-11-28:
Hi @Optimus1072 , how are you getting your images in NCHW format (with mean subtraction applied if your network model expects that)?

Since your batch size=1 works, but the other images do not, perhaps the other images are not getting into buffer memory correctly.

Optimus1072, 2018-11-28:
Hi,
Thanks for replying. I have tested thoroughly that my input buffer is in the required shape and format.

Optimus1072, 2018-11-28:
```
for(uint32_t bt = 0; bt < batch_size; ++bt)
    {
     
        cv::Mat sample_float;
        inp_frame[bt].convertTo(sample_float, CV_32FC3);
        // cv::Mat sample_normalized(INPUT_H, INPUT_W, CV_32FC3);
        // cv::subtract(sample_float, mean_, sample_normalized);
    
        // bool _rescaleTo01 = true;
        
        // if (_rescaleTo01)
        //     sample_normalized = sample_normalized * 0.007843;
        
        float* img_ptr = (float*)sample_float.data;
        // copy to data buffer
        
        const uint32_t imgPixels = height * width;
     
     
        for(uint32_t y = 0; y < height; ++y)
        {
            for(uint32_t x = 0; x < width; ++x)
            {
                // data[ (imgPixels * 0 + y * width + x + bt * imgPixels * 3) ] = (float)(((float)(*img_ptr) - 127.5f) * 0.007843);
                // data[ (imgPixels * 1 + y * width + x + bt * imgPixels * 3) ] = (float)(((float)(*(img_ptr + 1)) - 127.5f) * 0.007843);
                // data[ (imgPixels * 2 + y * width + x + bt * imgPixels * 3) ] = (float)(((float)(*(img_ptr + 2)) - 127.5f) * 0.007843);           
                
                data[ imgPixels * (bt * 3 + 0) + y * width + x ] = (float)(((float)(*img_ptr) - 127.5f) * 0.007843);
                data[ imgPixels * (bt * 3 + 1) + y * width + x ] = (float)(((float)(*(img_ptr + 1)) - 127.5f) * 0.007843);
                data[ imgPixels * (bt * 3 + 2) + y * width + x ] = (float)(((float)(*(img_ptr + 2)) - 127.5f) * 0.007843);
                
                img_ptr += 3;  
            }
        }

    }
```

This is how I am populating my input buffer

csbenk, 2019-01-08:
Hey @Optimus1072 could you get the batch inferencing done? Did you find out any other work around for the wrong outputs?

Optimus1072, 2019-01-09:
No I coudn't find any workaround

Optimus1072, 2019-01-10:
Hi, I have tried both ways and it doesn't give correct output. Were you able to make it work ?.

csbenk, 2019-01-11:
Nope. Still trying!

csbenk, 2019-01-11:
As far as I understand, increasing the batchsize in enqueue ensures it gives multiple outputs. But the image input is not taken in my case, whatever the image I pass as second image, the output is not changing, it's the same even if I don't give second image.

dusty-nv, 2019-01-11:
Sorry that you are still having trouble.  Are you increasing IBuilder::setMaxBatchSize() and IBuilder::setMaxWorkspaceSize() accordingly?

Since this is different code from this GitHub project, you may find a related question or more assistance on the TensorRT forum here:  [https://devtalk.nvidia.com/default/board/304/tensorrt/](https://devtalk.nvidia.com/default/board/304/tensorrt/)

csbenk, 2019-01-16:
Hey @dusty-nv , I feel that we need to change in the following function:
```
cudaError_t cudaPreImageNetMean( float* input, size_t inputWidth, size_t inputHeight,
	float* output, size_t outputWidth, size_t outputHeight, 
  const float& mean_value, cudaStream_t stream ){
......
}

```
This is where the preprocess happens and where first image is copied to gpu stream. The provision to copy second image is not present or atleast is not straight forward for me. Can you clarify if this function takes care of batch-size or we need to manipulate to make it work?

The above function is present in `imageNet.cu` file.

Optimus1072, 2019-01-17:
@csbenk Thanks for your suggestion, but Unfortunately, I wrote my own function for  this Pre-preocessing step and I can't find any fault with that.

Optimus1072, 2019-01-17:
@csbenk Are you trying to run some example or have you written your  own layers using IPluginImplement class?. Because for examples I was getting proper result along with speed improvement using multi-batch inference.

closed 2023-04-27 by dusty-nv