dusty-nv/jetson-inference #281: fatal error: NvInfer.h: No such file or directory

jacobswong, 2018-11-28:
I follow the step "Compiling the Project"  but get
~/jetson-inference/build$ make
[  2%] Building CXX object CMakeFiles/jetson-inference.dir/segNet.cpp.o
In file included from /home/jacob/jetson-inference/segNet.h:27:0,
                 from /home/jacob/jetson-inference/segNet.cpp:23:
/home/jacob/jetson-inference/tensorNet.h:27:21: fatal error: NvInfer.h: No such file or directory
 #include "NvInfer.h"
                     ^
compilation terminated.
CMakeFiles/jetson-inference.dir/build.make:1409: recipe for target 'CMakeFiles/jetson-inference.dir/segNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/segNet.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

I have tried to download TensorRT 5 again but does not solve the problem.  I search the system and do not find NvInfer.h

Jacob

dusty-nv, 2018-11-28:
@jacobswong what platform are you running on?

It appears that TensorRT has not been installed successfully.

jacobswong, 2018-11-28:
I did the following.  Do I miss something?

 sudo apt install ./nv-tensorrt-repo-ubuntu1604-cuda10.0-trt5.0.2.6-ga-20181009_1-1_amd64.deb [sudo] password for jacob: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'nv-tensorrt-repo-ubuntu1604-cuda10.0-trt5.0.2.6-ga-20181009' instead of './nv-tensorrt-repo-ubuntu1604-cuda10.0-trt5.0.2.6-ga-20181009_1-1_amd64.deb'
The following NEW packages will be installed:
  nv-tensorrt-repo-ubuntu1604-cuda10.0-trt5.0.2.6-ga-20181009
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/832 MB of archives.
After this operation, 832 MB of additional disk space will be used.
Get:1 /home/jacob/Downloads/nv-tensorrt-repo-ubuntu1604-cuda10.0-trt5.0.2.6-ga-20181009_1-1_amd64.deb nv-tensorrt-repo-ubuntu1604-cuda10.0-trt5.0.2.6-ga-20181009 amd64 1-1 [832 MB]
Selecting previously unselected package nv-tensorrt-repo-ubuntu1604-cuda10.0-trt5.0.2.6-ga-20181009.
(Reading database ... 253533 files and directories currently installed.)
Preparing to unpack .../nv-tensorrt-repo-ubuntu1604-cuda10.0-trt5.0.2.6-ga-20181009_1-1_amd64.deb ...
Unpacking nv-tensorrt-repo-ubuntu1604-cuda10.0-trt5.0.2.6-ga-20181009 (1-1) ...
Setting up nv-tensorrt-repo-ubuntu1604-cuda10.0-trt5.0.2.6-ga-20181009 (1-1) ...

dusty-nv, 2018-11-28:
I am not exactly sure about the install procedure of TensorRT on PC, as I am used to Jetson with JetPack.

Can you run this:

```sudo apt-get update
apt-cache search tensorrt```

It might be that you installed this "repo" package, but still need to install the individual TensorRT packages which are provided by the repo package.

ailuropoda0, 2019-01-10:
Did you install libnvinfer?
You can check with the command `dpkg -l | grep nvinfer`
If nothing shows up, you should install libnvinfer

mc-android-developer, 2019-03-28:
sudo aptitude install libnvinfer5 libnvinfer-dev

dc3671, 2019-08-09:
If you are `Ubuntu 16`, `libnvinfer5` and `libnvinfer-dev` are not available directly. You need to add nvidia's cuda-repo and machine-learning-repo.
```
sudo apt-get install gnupg-curl
wget https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1604/x86_64/cuda-repo-ubuntu1604_10.0.130-1_amd64.deb
sudo dpkg -i cuda-repo-ubuntu1604_10.0.130-1_amd64.deb
sudo apt-key adv --fetch-keys https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1604/x86_64/7fa2af80.pub
sudo apt-get update
wget http://developer.download.nvidia.com/compute/machine-learning/repos/ubuntu1604/x86_64/nvidia-machine-learning-repo-ubuntu1604_1.0.0-1_amd64.deb
sudo apt install ./nvidia-machine-learning-repo-ubuntu1604_1.0.0-1_amd64.deb
sudo apt-get update
```

Reference: https://www.tensorflow.org/install/gpu#ubuntu_1604_cuda_10

wangzhankun, 2020-04-19:
> Did you install libnvinfer?
> You can check with the command `dpkg -l | grep nvinfer`
> If nothing shows up, you should install libnvinfer

I have nvinfer installed. But either not found NvInfer.h. I run the ```find``` command, but find nothing.
```Bash
find / NvInfer.h
```
```
ii  libnvinfer-bin                             6.0.1-1+cuda10.0                                 arm64        TensorRT binaries
ii  libnvinfer-dev                             6.0.1-1+cuda10.0                                 arm64        TensorRT development libraries and headers
ii  libnvinfer-doc                             6.0.1-1+cuda10.0                                 all          TensorRT documentation
ii  libnvinfer-plugin-dev                      6.0.1-1+cuda10.0                                 arm64        TensorRT plugin libraries
ii  libnvinfer-plugin6                         6.0.1-1+cuda10.0                                 arm64        TensorRT plugin libraries
ii  libnvinfer-samples                         6.0.1-1+cuda10.0                                 all          TensorRT samples
ii  libnvinfer6                                6.0.1-1+cuda10.0                                 arm64        TensorRT runtime libraries
ii  python3-libnvinfer                         6.0.1-1+cuda10.0                                 arm64        Python 3 bindings for TensorRT
ii  python3-libnvinfer-dev                     6.0.1-1+cuda10.0                                 arm64        Python 3 development package for TensorRT

```
```
ii  nvidia-container-csv-tensorrt              6.0.1.10-1+cuda10.0                              arm64        Jetpack TensorRT CSV file
ii  tensorrt                                   6.0.1.10-1+cuda10.0                              arm64        Meta package of TensorRT

```
## OS
* jetson nano ubuntu18.04
* cuda 10.0

tlhcelik, 2020-06-03:
**sudo apt-get install tensorrt** 
it's work me

dusty-nv, 2021-10-08:
Can you try `sudo apt-get install nvidia-tensorrt libnvinfer-dev`

gabegarz, 2021-11-11:
Hi all, I have an old Ubuntu 18 machine with an old GPU.  Trying to install TensorRt 9.0 with driver 390.144 and Cuda 9.0.85.  When I get to install I get an  error   
"tensorrt : Depends: libnvinfer8 (= 8.2.0-1+cuda11.4) but it is not going to be installed"  

I have researched this a lot and everything point to TensorRT but I don't know enough about it to solve this.

closed 2023-04-27 by dusty-nv