dusty-nv/jetson-inference #282: error: ‘Dims3’ is not a member of ‘nvinfer1’

robotlinker, 2018-12-05:
Hi All,

I got a make problem no the Jetson Tx2, any idea?

nvidia@tegra-ubuntu:~/jetson-inference/build$ make
[  1%] Linking CXX shared library ../aarch64/lib/libjetson-utils.so
[ 50%] Built target jetson-utils
[ 51%] Building CXX object CMakeFiles/jetson-inference.dir/tensorNet.cpp.o
In file included from /home/nvidia/jetson-inference/tensorNet.cpp:24:0:
/home/nvidia/jetson-inference/build/aarch64/include/randInt8Calibrator.h:45:34: error: ‘Dims3’ is not a member of ‘nvinfer1’
      const std::map<std::string, nvinfer1::Dims3>& inputDimensions );
                                  ^
/home/nvidia/jetson-inference/build/aarch64/include/randInt8Calibrator.h:45:34: note: suggested alternative:
In file included from /home/nvidia/jetson-inference/tensorNet.cpp:23:0:
/home/nvidia/jetson-inference/tensorNet.h:35:27: note:   ‘Dims3’
 typedef nvinfer1::DimsCHW Dims3;
                           ^
In file included from /home/nvidia/jetson-inference/tensorNet.cpp:24:0:
/home/nvidia/jetson-inference/build/aarch64/include/randInt8Calibrator.h:45:34: error: ‘Dims3’ is not a member of ‘nvinfer1’
      const std::map<std::string, nvinfer1::Dims3>& inputDimensions );
                                  ^
/home/nvidia/jetson-inference/build/aarch64/include/randInt8Calibrator.h:45:34: note: suggested alternative:
In file included from /home/nvidia/jetson-inference/tensorNet.cpp:23:0:
/home/nvidia/jetson-inference/tensorNet.h:35:27: note:   ‘Dims3’
 typedef nvinfer1::DimsCHW Dims3;
                           ^
In file included from /home/nvidia/jetson-inference/tensorNet.cpp:24:0:
/home/nvidia/jetson-inference/build/aarch64/include/randInt8Calibrator.h:45:49: error: template argument 2 is invalid
      const std::map<std::string, nvinfer1::Dims3>& inputDimensions );
                                                 ^
/home/nvidia/jetson-inference/build/aarch64/include/randInt8Calibrator.h:45:49: error: template argument 4 is invalid
/home/nvidia/jetson-inference/build/aarch64/include/randInt8Calibrator.h:78:24: error: ‘Dims3’ is not a member of ‘nvinfer1’
  std::map<std::string, nvinfer1::Dims3> mInputDimensions;

dusty-nv, 2018-12-05:
Hi @RobotLinker,  are you using JetPack 3.3?  If not, which version?

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

robotlinker, 2018-12-05:
Here it is :+1: 
nvidia@tegra-ubuntu:~/jetson-inference/build$ dpkg -l | grep nvinfer
ii  libnvinfer-dev                                    3.0.2-1+cuda8.0                                       arm64        TensorRT development libraries and headers
ii  libnvinfer3                                       3.0.2-1+cuda8.0                                       arm64        TensorRT runtime libraries

varun-tangoit, 2018-12-05:
I'm also facing same issue, can you give some suggestion how do i resolve this issue.

dusty-nv, 2018-12-05:
Hi guys, just pushed a fix for JetPack 3.2 to master in commit [`f23ea0`](https://github.com/dusty-nv/jetson-inference/commit/f23ea01058617bd55643098357b60159784e71ff).  Can you try cloning the repo again and re-building?

robotlinker, 2018-12-06:
Hi @dusty-nv , it works now, thank you!

closed 2018-12-06 by robotlinker