dusty-nv/jetson-inference #285: Segmentation fault while running segNet on video file

adithya-p, 2018-12-25:
The dimension of the video is: 1280x720
Streaming video through OpenCV, The following snippet of code does the job of copying Mat to Cuda memory for further processing:

```
cudaMemcpy2D((void*)imgBufferRGB, imgW*sizeof(uchar3), (void*)frame.data, frame.step, imgW*sizeof(uchar3), imgH, cudaMemcpyHostToDevice);
cudaRGB8ToRGBA32(imgBufferRGB, imgBufferRGBAf, imgW, imgH);
if( !net->Overlay((float*)imgBufferRGBAf, (float*)outCUDA, imgW, imgH) )
		{
			printf("segnet-console:  failed to process segmentation overlay.\n");
			continue;
		} 
```
Didn't make any changes to the source files except the above few lines. 

The log is as follows:

```
segnet-video
  args (2):  0 [./segnet-video]  1 [/home/common/NYC.mp4]  

segNet -- loading segmentation network model from:
       -- prototxt:   networks/FCN-Alexnet-Cityscapes-HD/deploy.prototxt
       -- model:      networks/FCN-Alexnet-Cityscapes-HD/snapshot_iter_367568.caffemodel
       -- labels:     networks/FCN-Alexnet-Cityscapes-HD/cityscapes-labels.txt
       -- colors:     networks/FCN-Alexnet-Cityscapes-HD/cityscapes-deploy-colors.txt
       -- input_blob  'data'
       -- output_blob 'score_fr_21classes'
       -- batch_size  2

[TRT]  TensorRT version 5.0.2
[TRT]  desired precision specified for GPU: FASTEST
[TRT]  requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]  native precisions detected for GPU:  FP32, INT8
[TRT]  selecting fastest native precision for GPU:  FP32
[TRT]  attempting to open engine cache file networks/FCN-Alexnet-Cityscapes-HD/snapshot_iter_367568.caffemodel.2.1.GPU.FP32.engine
[TRT]  loading network profile from engine cache... networks/FCN-Alexnet-Cityscapes-HD/snapshot_iter_367568.caffemodel.2.1.GPU.FP32.engine
[TRT]  device GPU, networks/FCN-Alexnet-Cityscapes-HD/snapshot_iter_367568.caffemodel loaded
[TRT]  device GPU, CUDA engine context initialized with 2 bindings
[TRT]  networks/FCN-Alexnet-Cityscapes-HD/snapshot_iter_367568.caffemodel input  binding index:  0
[TRT]  networks/FCN-Alexnet-Cityscapes-HD/snapshot_iter_367568.caffemodel input  dims (b=2 c=3 h=1024 w=2048) size=50331648
[cuda]  cudaAllocMapped 50331648 bytes, CPU 0x7f75f0000000 GPU 0x7f75f0000000
[TRT]  networks/FCN-Alexnet-Cityscapes-HD/snapshot_iter_367568.caffemodel output 0 score_fr_21classes  binding index:  1
[TRT]  networks/FCN-Alexnet-Cityscapes-HD/snapshot_iter_367568.caffemodel output 0 score_fr_21classes  dims (b=2 c=21 h=26 w=58) size=253344
[cuda]  cudaAllocMapped 253344 bytes, CPU 0x7f75f3000000 GPU 0x7f75f3000000
device GPU, networks/FCN-Alexnet-Cityscapes-HD/snapshot_iter_367568.caffemodel initialized.
[cuda]  cudaAllocMapped 336 bytes, CPU 0x7f75c3400200 GPU 0x7f75c3400200
[TRT]  segNet outputs -- s_w 58  s_h 26  s_c 21
[cuda]  cudaAllocMapped 1508 bytes, CPU 0x7f75f3200000 GPU 0x7f75f3200000
segNet -- class 00  color 0 0 0 255
segNet -- class 01  color 111 74 0 255
segNet -- class 02  color 81 0 81 255
segNet -- class 03  color 128 64 128 180
segNet -- class 04  color 244 35 232 255
segNet -- class 05  color 250 170 160 255
segNet -- class 06  color 70 70 70 220
segNet -- class 07  color 102 102 156 255
segNet -- class 08  color 190 153 153 255
segNet -- class 09  color 180 165 180 255
segNet -- class 10  color 150 100 100 255
segNet -- class 11  color 153 153 153 150
segNet -- class 12  color 250 170 30 255
segNet -- class 13  color 220 220 0 255
segNet -- class 14  color 107 142 35 255
segNet -- class 15  color 152 251 152 255
segNet -- class 16  color 70 130 180 254
segNet -- class 17  color 220 20 60 255
segNet -- class 18  color 0 0 142 255
segNet -- class 19  color 0 0 70 255
segNet -- class 20  color 119 11 32 255
segNet -- loaded 21 class colors
segNet -- class 00  label 'void'
segNet -- class 01  label 'dynamic'
segNet -- class 02  label 'ground'
segNet -- class 03  label 'road'
segNet -- class 04  label 'sidewalk'
segNet -- class 05  label 'parking'
segNet -- class 06  label 'building'
segNet -- class 07  label 'wall'
segNet -- class 08  label 'fence'
segNet -- class 09  label 'guard rail'
segNet -- class 10  label 'bridge tunnel'
segNet -- class 11  label 'pole'
segNet -- class 12  label 'traffic light'
segNet -- class 13  label 'traffic sign'
segNet -- class 14  label 'vegetation'
segNet -- class 15  label 'terrain'
segNet -- class 16  label 'sky'
segNet -- class 17  label 'person'
segNet -- class 18  label 'car'
segNet -- class 19  label 'truck'
segNet -- class 20  label 'cycle'
segNet -- loaded 21 class labels
[cuda]  cudaAllocMapped 14745600 bytes, CPU 0x7f75d8800000 GPU 0x7f75d8800000
default X screen 0:   1920 x 1080
[OpenGL]  glDisplay display window initialized
[OpenGL]   creating 1280x720 texture

segnet-video:  video open for streaming
[TRT]  segNet::Overlay -- s_w 58  s_h 26  s_c 21  s_x 0.028320  s_y 0.025391
[TRT]  segNet::Overlay -- ignoring class 'void' id=0
Segmentation fault
```
The issue is at s[egNet.cpp#L531-534](https://github.com/dusty-nv/jetson-inference/blob/master/segNet.cpp#L531-L534)

Can anyone shed some light on the problem? 
CC: @dusty-nv

adithya-p, 2019-02-01:
The issue was the way in which the OpenCV `Mat` is being copied to the CUDA Memory. Modified it to use [ConvertRGBA](https://github.com/dusty-nv/jetson-utils/blob/b337733d4f1000b87979cb982ced30445d42b5b3/camera/gstCamera.cpp#L93) function and It worked.

closed 2019-02-01 by adithya-p

aviontics, 2019-08-14:
'adithya' please can you share code snippet

adithya-p, 2019-08-14:
@aviontics Here's the link to the source code - [segnet-camera.cpp](https://github.com/dusty-nv/jetson-inference/blob/master/examples/segnet-camera/segnet-camera.cpp)