dusty-nv/jetson-inference #286: Can I use Ubuntu 16 System on Jetson Xavier?

zhearing, 2018-12-25:
my program can only run on ubuntu16. How does Xavier brush the machine to ubuntu16?

Sprinzzzz, 2019-01-30:
```
./imagenet-console orange_0.jpg output_0.jpg
./imagenet-console: error while loading shared libraries: libjetson-inference.so: cannot open shared object file: No such file or directory
```
I tried to classify images using console program on jetson Xavier. Please help me with a solution. Note: I hooked Jetson xavier to a common network at office

dusty-nv, 2019-04-09:
Hi @zhearing, Jetson AGX Xavier (JetPack 4.1 and newer) supports Ubuntu 18.04.   Jetson TX2 and Jetson Nano also have now moved to Ubuntu 18.04 with JetPack 4.2.

Durant35, 2019-11-29:
@dusty-nv, thus could we switch to Ubuntu16.04 on Jetson AGX Xavier? If so, how?

closed 2023-04-27 by dusty-nv