dusty-nv/jetson-inference #287: how can i use the SegNet to be based on Alexnet

asmaa4may, 2019-01-15:


dusty-nv, 2019-01-15:
> -----------------------------------------------------------------------------------
> This email message is for the sole use of the intended recipient(s) and may contain
> confidential information.  Any unauthorized review, use, disclosure or distribution
> is prohibited.  If you are not the intended recipient, please contact the sender by
> reply email and destroy all copies of the original message.
> -----------------------------------------------------------------------------------

The pretrained built-in segNet models that come with the repo are based on Alexnet (fully convolutional).

asmaa4may, 2019-01-16:
> The pretrained built-in segNet models that come with the repo are based on Alexnet (fully convolutional).
> […](#)
> ----------------------------------------------------------------------------------- This email message is for the sole use of the intended recipient(s) and may contain confidential information. Any unauthorized review, use, disclosure or distribution is prohibited. If you are not the intended recipient, please contact the sender by reply email and destroy all copies of the original message. -----------------------------------------------------------------------------------
sir you mean that SegNet already based on Alexnet. because the code within matlab for segnet is
lgraph = segnetLayers(imageSize,numClasses,'model');
and here model must be vgg16 or vgg19 as the documentation say

closed 2023-04-27 by dusty-nv