dusty-nv/jetson-inference #293: TX1 tensorNet.cpp.o Error 1

ismailbaayer, 2019-02-22:
Hello,

 I have tried to went through the tutorial. I have tried to build a source on Jetson. My jetson runs Ubuntu 16.04. I have followed the instructions; however when I run "make" on terminal, I have received following error:
`Scanning dependencies of target jetson-inference
[ 53%] Building CXX object CMakeFiles/jetson-inference.dir/tensorNet.cpp.o
In file included from /home/nvidia/jetson-inference/tensorNet.cpp:23:0:
/home/nvidia/jetson-inference/tensorNet.h:135:18: error: ‘nvinfer1::IInt8Calibrator’ has not been declared
        nvinfer1::IInt8Calibrator* calibrator=NULL, cudaStream_t stream=NULL );
                  ^
/home/nvidia/jetson-inference/tensorNet.h:150:18: error: ‘nvinfer1::IInt8Calibrator’ has not been declared
        nvinfer1::IInt8Calibrator* calibrator=NULL, cudaStream_t stream=NULL );
                  ^
/home/nvidia/jetson-inference/tensorNet.h:247:19: error: ‘nvinfer1::IInt8Calibrator’ has not been declared
         nvinfer1::IInt8Calibrator* calibrator, std::ostream& modelStream);
                   ^
/home/nvidia/jetson-inference/tensorNet.cpp: In function ‘nvinfer1::DataType precisionTypeToTRT(precisionType)’:
/home/nvidia/jetson-inference/tensorNet.cpp:75:26: error: ‘kINT8’ is not a member of ‘nvinfer1::DataType’
   case TYPE_INT8: return nvinfer1::DataType::kINT8;
                          ^
/home/nvidia/jetson-inference/tensorNet.cpp: At global scope:
/home/nvidia/jetson-inference/tensorNet.cpp:297:20: error: ‘nvinfer1::IInt8Calibrator’ has not been declared
          nvinfer1::IInt8Calibrator* calibrator,  
                    ^
/home/nvidia/jetson-inference/tensorNet.cpp: In member function ‘bool tensorNet::ProfileModel(const string&, const string&, const std::vector<std::__cxx11::basic_string<char> >&, unsigned int, precisionType, deviceType, bool, int*, std::ostream&)’:
/home/nvidia/jetson-inference/tensorNet.cpp:421:86: error: ‘class nvinfer1::IBuilder’ has no member named ‘getInt8Mode’
 device %s, building INT8:  %s\n", deviceTypeToStr(device), builder->getInt8Mode
                                                                     ^
/home/nvidia/jetson-inference/tensorNet.cpp: At global scope:
/home/nvidia/jetson-inference/tensorNet.cpp:463:20: error: ‘nvinfer1::IInt8Calibrator’ has not been declared
          nvinfer1::IInt8Calibrator* calibrator, cudaStream_t stream )
                    ^
/home/nvidia/jetson-inference/tensorNet.cpp:477:20: error: ‘nvinfer1::IInt8Calibrator’ has not been declared
          nvinfer1::IInt8Calibrator* calibrator, cudaStream_t stream )
                    ^
/home/nvidia/jetson-inference/tensorNet.cpp: In member function ‘bool tensorNet::LoadNetwork(const char*, const char*, const char*, const char*, const std::vector<std::__cxx11::basic_string<char> >&, uint32_t, precisionType, deviceType, bool, int*, cudaStream_t)’:
/home/nvidia/jetson-inference/tensorNet.cpp:482:86: error: ‘NV_TENSORRT_PATCH’ was not declared in this scope
 TensorRT version %u.%u.%u\n", NV_TENSORRT_MAJOR, NV_TENSORRT_MINOR, NV_TENSORRT
                                                                     ^
CMakeFiles/jetson-inference.dir/build.make:76: recipe for target 'CMakeFiles/jetson-inference.dir/tensorNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/tensorNet.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
`
Could you please help me to solve this problem?
Do I have to build a host before building the source on Jetson?

Looking forward to hearing from you

dusty-nv, 2019-02-22:
Hi, which version of JetPack is your Jetson running?

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

Joaquin1910, 2019-03-05:
Hello dusty. I have the same problem. Please help me, I need this for my thesis. Thank you

dusty-nv, 2019-03-06:
Hi @Joaquin1910 , @ismailbaayer , can you please pull from master branch and try compiling again?

I applied some fixes in commit [`835447`](https://github.com/dusty-nv/jetson-inference/commit/835447bdade514ec066615db3ec9bcd60792e8cb).

Joaquin1910, 2019-03-13:
Hi @dusty-nv and @ismailbaayer .it was fixed, when installing jetpack, tensorrt was not installed, so, install it and it worked.

Mr @dusty-nv . I am following the github jetson-inference, and I would like to know how I can adapt the code so that the acquisition is through an IP camera, that is to say that the image comes via Ethernet, if you have an example I would thank you. On the other hand, I would like to know how I can perform a profiling on the performance of the inference to see metrics between the CPU and the GPU. It's for a colombia school project. I thank you for your help. Thank you

dusty-nv, 2019-03-13:
Hi, see this fork for IP camera:   https://github.com/Abaco-Systems/jetson-inference-gv

ismailbaayer, 2019-03-13:
Hi thanks it works now :)

Joaquin1910, 2019-03-16:
Hi @dusty-nv 
this only applies only to abacus IP cameras or also to another type of camera

closed 2023-04-27 by dusty-nv