dusty-nv/jetson-inference #294: Where can I find the datasets for ped-100, multiped-500 and Facenet-120?

Sprinzzzz, 2019-03-06:
I am new down here and following the tutorial. I got stuck with the place of finding the datasets. I could find the coco datasets as I downloaded as mentioned in the article. Is there a way by which I could obtain the other datasets ?

Thanks in advance

dusty-nv, 2019-03-06:
Hi, the `facenet` model was trained with FDDB dataset:   http://vis-www.cs.umass.edu/fddb/

The dataset for the pedestrian detectors isn't public, however there are open datasets for pedestrian detection (like KITTI and COCO)

Sprinzzzz, 2019-03-06:
> Hi, the `facenet` model was trained with FDDB dataset: http://vis-www.cs.umass.edu/fddb/
> 
> The dataset for the pedestrian detectors isn't public, however there are open datasets for pedestrian detection (like KITTI and COCO)

Thanks a lot dusty!!

closed 2019-03-07 by Sprinzzzz