dusty-nv/jetson-inference #300: Hard lockup Jetson Nano imagenet-console

NHDLarry, 2019-03-25:
Cloned and built on Jetson Nano the jetson-inference project. Added shared library path (see other issue filed).  Tried to run the example:

$ ./imagenet-console orange_0.jpg output_0.jpg 

Made it to:  "[TRT]  device GPU, completed building CUDA engine"

The Nano basically hard-locked the board, requiring a hard reset after several minutes. Some mouse movement intermittently. No other signs of life, no keystrokes, nothing. Let it go 5 minutes and hard reset. 

After hard reboot, attempted again and it made it to the same point, "[TRT]  device GPU, completed building CUDA engine"  However, board was alive, window dragging and other tasks worked. After several minutes, the program produced the expected output with a score of 97.852%. 

Seems like a really long delay in building the CUDA engine step.

dusty-nv, 2019-04-09:
Hi @NHDLarry, it can typically take several minutes to build the CUDA engine with TensorRT.  During this time, it is performing optimizations and microkernel tuning on the platform.  If you run `sudo jetson_clocks` beforehand, this may help.

If you keep experience hard lockups, it may be related to your power supply.  See [this thread](https://devtalk.nvidia.com/default/topic/1048640/jetson-nano/power-supply-considerations-for-jetson-nano-developer-kit/) for recommended power supplies, or put your Nano into 5W mode by running the following:

```bash
$ sudo npvmodel -m 1
```

calocedrus, 2019-05-22:
@dusty-nv setting the power to 5W mode did help: I've eventually got the output in about 5mn, while in 10W mode the board went into 3 hard lockups and 2 sudden switch off.
And this was with a benchtop power supply able to throw up to 10A and set at 5.3V to compensate for the voltage drop at high currents over a micro USB cable...

By the way, it's `nvpmodel` and not `npvmodel`

I'll try to hook up a jack barrel.
In any case this clearly seem to be power related.

pringithub, 2019-05-29:
I ran ./imagenet-console <yaddayadda> and it worked fine. 
Now, I'm getting a complete shutdown (no lockup) when attempting to run ./segnet-console ...
Probably also related to this.,,
How does running `$ sudo npvmodel -m 1` put the Nano into 5W mode? What is the nominal power setting? How would I switch back?

closed 2023-04-27 by dusty-nv