dusty-nv/jetson-inference #305: missing deep_homography_webcam_320.onnx

lu-hongwei, 2019-04-03:
the file 'https://nvidia.box.com/shared/static/nlqbsdnt76y0nmkwdzxkg4zbvhk4bidh.gz' Deep-Homography-COCO.tar.gz does  not contain "deep_homography_webcam_320.onnx " as was mentioned in homography-camera.cpp when networkType == WEBCAM_320.

kozak-yakym, 2019-06-13:
I've ran with the `networks/Deep-Homography-COCO/deep_homography.onnx` model:

>$ ./homography-camera --model='coco'

But before I fixed DEFAULT_CAMERA macro to -1 to run on my JetsonTX2 on-board camera.
>homography-camera.cpp 
>#define DEFAULT_CAMERA -1

vvzvlad, 2019-08-28:
> Same here.
> I've ran only with the `networks/Deep-Homography-COCO/deep_homography.onnx` model:
> 
> > $ ./homography-camera --model='coco'
> 
> But before I fixed DEFAULT_CAMERA macro to -1 to run on my JetsonTX2 on-board camera.
> 
> > homography-camera.cpp
> > #define DEFAULT_CAMERA -1

```
[TRT]   /home/erisuser/p4sw/sw/gpgpu/MachineLearning/DIT/release/5.1/parsers/onnxOpenSource/builtin_op_importers.cpp:836: Convolution output dimensions: (128, 16, 16)
[TRT]   90:Conv -> (128, 16, 16)
[TRT]   91:Relu -> (128, 16, 16)
[TRT]   92:Constant -> (2)
[TRT]   93:Reshape -> (32768)
[TRT]   94:Dropout -> (32768)
[TRT]   /home/erisuser/p4sw/sw/gpgpu/MachineLearning/DIT/release/5.1/parsers/onnxOpenSource/builtin_op_importers.cpp:1981: Unsqueezing from (32768) to (32768, 16)
While parsing node number 29 [Unsqueeze]:
ERROR: builtin_op_importers.cpp:1987 In function importUnsqueeze:
[8] Assertion failed: get_shape_size(layer->getOutput(0)->getDimensions()) == get_shape_size(old_shape)
[TRT]   failed to parse ONNX model '/usr/local/bin/networks/Deep-Homography-COCO/deep_homography.onnx'
[TRT]   device GPU, failed to load networks/Deep-Homography-COCO/deep_homography.onnx
[TRT]   failed to load homographyNet
homography-camera:   failed to initialize homographyNet
```

closed 2023-04-27 by dusty-nv