dusty-nv/jetson-inference #307: (un)Tar of models fails in cmake. Models have Windows special charaters in their names.

Rmwak, 2019-04-04:
After the models are downloaded in the CMake process, for example as "Super-Resolution-BSD500.tar.gz", the tar step fails:

```
XXXXXX - ´ Super-Resolution-BSD500.tar.gz\r´ saved [640910/640910]

tar (child): Super-Resolution-BSD500.tar.gz: Cannot open: No such file or directory.
tar (child): Error is not recoverable: exiting now
tar: Child returned status 2
```
Seems that somehow the filename got encoded with Windows line endings '\r'.

Other examples are:
'multiped-500.tar.gz'$'\r'
'facenet-120.tar.g'$'\r'
'googlenet_noprob.prototxt'$'\r'

Not terrible, since the cmake and make scripts dont' really fail and you can actually extract those files with the correct name and move them to the networks directory, but is still an error.

dusty-nv, 2019-04-09:
Hmm, I just checked the file from Ubuntu with `cat -e CMakePreBuild.sh`, and it only showed Unix line-endings (`$`) and not Windows line endings (`^M$`).  

Maybe it was a temporary glitch from a previous version of the file.  Can you try cloning from master again and seeing if the issue persists?

Rmwak, 2019-04-11:
Apologies, I think I didn't explain myself correctly. The issue wasn't with the CMake script, but with the name of the models stored at public.boxcloud.com which are downloaded. Those were the ones with windows special characters in their name, so the tar steps at the `CMakePreBuild.sh` failed.

In any case, just tested it again and the issue is gone. I guess someone updated them?  I'll close the issue now.

Thanks Dusty

closed 2019-04-11 by Rmwak