dusty-nv/jetson-inference #308: Segmentation fault  running imagenet-console on jetson nano

IIMKB, 2019-04-09:
I am trying to run  the jetson-inference samples on jetson nano.  Running the imagenet-console test app results in the following error:

[TRT]  device GPU, building CUDA engine (this may take a few minutes the first time a network is loaded)
Segmentation fault

`
$ sudo ./imagenet-console orange_0.jpg output_0.jpg
imagenet-console
  args (3):  0 [./imagenet-console]  1 [orange_0.jpg]  2 [output_0.jpg]  

imageNet -- loading classification network model from:
         -- prototxt     networks/googlenet.prototxt
         -- model        networks/bvlc_googlenet.caffemodel
         -- class_labels networks/ilsvrc12_synset_words.txt
         -- input_blob   'data'
         -- output_blob  'prob'
         -- batch_size   2

[TRT]  TensorRT version 5.0.6
[TRT]  detected model format - caffe  (extension '.caffemodel')
[TRT]  desired precision specified for GPU: FASTEST
[TRT]  requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]  native precisions detected for GPU:  FP32, FP16
[TRT]  selecting fastest native precision for GPU:  FP16
[TRT]  attempting to open engine cache file networks/bvlc_googlenet.caffemodel.2.1.GPU.FP16.engine
[TRT]  cache file not found, profiling network model on device GPU
[TRT]  device GPU, loading networks/googlenet.prototxt networks/bvlc_googlenet.caffemodel
[TRT]  failed to retrieve tensor for Output "prob"
[TRT]  device GPU, configuring CUDA engine
[TRT]  device GPU, building FP16:  ON
[TRT]  device GPU, building INT8:  OFF
[TRT]  device GPU, building CUDA engine (this may take a few minutes the first time a network is loaded)
Segmentation fault
`
Can someone please suggest me what all things i can solve this issues

IIMKB, 2019-04-10:
Thanks for replying.
I have downloaded the new  `bvlc_googlenet.caffemodel`.
But the `Segmentation faul`t still occurs. 
Can you give me some other suggestions?
Thank you very much for your help.

dusty-nv, 2019-04-11:
Hi @IIMKB, can you try the steps from this post?

https://devtalk.nvidia.com/default/topic/1049144/jetson-nano/imagenet-console-application-not-working-properly/post/5328776/#5328776

FredXu1990, 2019-04-14:
I have also met this problem, and I have tried the above solutions. However, it doesn't make sense, could you give me some other suggestions? Thanks a lot.

kaijunzhan, 2019-04-23:
I have similar issue but rather getting segmentation fault - nano just crashed. reinstalled / re-image still crash

liurongfang, 2019-04-23:
I have same problem,but it be resolved after re-download caffmodel.hope this can help you

hvacking, 2019-06-25:
hi ,in china, this url is not available. can i use https://github.com/dusty-nv/jetson-inference/releases to download the ' bvlc_googlenet.caffemodel '?

dusty-nv, 2019-06-25:
Yes, you can use that mirror URL.  Are you able to access the mirror URL from China?

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

hvacking, 2019-06-25:
yes ,i can access to mirror URL of github.
but when we proceed the install, all the process stopped at this step. And some essential fold is not exit,such as 'aarch64' folder.
how do i fix this problem?

[image]
[image]

dusty-nv, 2019-06-25:
Try running "cmake ../" again.  Then "make" and "sudo make install"

[11]<https://user-images.githubusercontent.com/28031945/60103924-0a482e00-9793-11e9-8bc6-2275721189fb.gif>
[333]<https://user-images.githubusercontent.com/28031945/60104004-2ea40a80-9793-11e9-9ce3-0caa7ec6eaae.gif>

-----------------------------------------------------------------------------------

hvacking, 2019-06-26:
hi ,i tried several times, but all stops at connect nivida.app.box.com ，when i interrupt the the process. i can not find the aarch64 folder under builder folder.

dusty-nv, 2019-06-26:
OK, go into CMakePreBuild.sh and comment out all the wget commands.  Then run cmake ../ and make.

martynoconnor, 2019-07-25:
I have the same same problem. If I call imagenet-console.py, I get "Segmentation fault (core dumped)" - this is from a clean install following the github instructions to the letter, using an image downloaded an hour ago.

dusty-nv, 2019-07-25:
@martynoconnor does your log look the same as [above](https://github.com/dusty-nv/jetson-inference/issues/308#issue-430811734)?

Were you able to download the models ok from the script?   If your log looks the same as above, it means your googlenet model is corrupted - so try deleting and re-downloading them.

sdcharle, 2019-09-30:
For me (also running on a nano), console works, but the camera version gives a Segmentation Fault:

[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvvconv0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvarguscamerasrc0
[gstreamer] gstreamer msg stream-start ==> pipeline0
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
Segmentation fault (core dumped)

closed 2023-04-27 by dusty-nv