dusty-nv/jetson-inference #309: make build error

Fariborzz, 2019-04-09:
Hi All

I am trying to setup jetson-inference on Jetson Nano. However, I am facing an error to build after making the repository without any error:

[44%] Linking CXX shared library ../aarch64/lib/libjetson-utils.so 
/usr/bin/ld: cannot find -IGL
collect2: errir: ld returned 1 exit status

Can anyone help me to fix this problem?

Thanks.

dusty-nv, 2019-04-09:
Hi @Fariborzz, can you check your libGL.so file with this command?

```bash
$ ls -ll /usr/lib/aarch64-linux-gnu/libGL.so*
```

The output should look like this:

```bash
lrwxrwxrwx 1 root root     14 Aug 15  2018 /usr/lib/aarch64-linux-gnu/libGL.so -> libGL.so.1.0.0
lrwxrwxrwx 1 root root     14 Aug 15  2018 /usr/lib/aarch64-linux-gnu/libGL.so.1 -> libGL.so.1.0.0
-rw-r--r-- 1 root root 972968 Aug 15  2018 /usr/lib/aarch64-linux-gnu/libGL.so.1.0.0
```
In particular, verify that your `libGL.so.1.0.0` library is also `972968` bytes in size.

If it appears differently, then your `libGL.so` must have gotten overwritten or corrupted, perhaps by another package that you installed.

In that case, you can use a fresh SD card image, or you can try restoring it from a copy of my working `libGL.so` that I have uploaded here:   https://nvidia.box.com/s/cuqracl8465p3uo0mhv4l5108mwaeuti

Fariborzz, 2019-04-09:
Thank you @dusty-nv . I use a fresh SD card image and my problem is solved.

closed 2019-04-09 by Fariborzz

silvermind20, 2019-06-12:
> Hi @Fariborzz, can you check your libGL.so file with this command?
> 
> ```shell
> $ ls -ll /usr/lib/aarch64-linux-gnu/libGL.so*
> ```
> 
> The output should look like this:
> 
> ```shell
> lrwxrwxrwx 1 root root     14 Aug 15  2018 /usr/lib/aarch64-linux-gnu/libGL.so -> libGL.so.1.0.0
> lrwxrwxrwx 1 root root     14 Aug 15  2018 /usr/lib/aarch64-linux-gnu/libGL.so.1 -> libGL.so.1.0.0
> -rw-r--r-- 1 root root 972968 Aug 15  2018 /usr/lib/aarch64-linux-gnu/libGL.so.1.0.0
> ```
> 
> In particular, verify that your `libGL.so.1.0.0` library is also `972968` bytes in size.
> 
> If it appears differently, then your `libGL.so` must have gotten overwritten or corrupted, perhaps by another package that you installed.
> 
> In that case, you can use a fresh SD card image, or you can try restoring it from a copy of my working `libGL.so` that I have uploaded here: https://nvidia.box.com/s/cuqracl8465p3uo0mhv4l5108mwaeuti

I have same issue and tried your recommendation and here is my results:
 ```shell
lrwxrwxrwx 1 root root     41 Jun 11 16:05 /usr/lib/aarch64-linux-gnu/libGL.so -> /usr/lib/aarch64-linux-gnu/tegra/libGL.so
lrwxrwxrwx 1 root root     14 May 10 05:17 /usr/lib/aarch64-linux-gnu/libGL.so.1 -> libGL.so.1.0.0
-rw-r--r-- 1 root root 972968 May 10 05:17 /usr/lib/aarch64-linux-gnu/libGL.so.1.0.0
 ```
Is there anyway to fix it without using a fresh SD card image? 
Also, how can I restore the libGL.so? I am pretty new in this!

EHB-1989, 2020-04-06:
For me installing the following pre-requisite packages resolved the issue 👍 
$ sudo apt-get install libgstreamer-plugins-base1.0-dev libgstreamer1.0-dev libgflags-dev