dusty-nv/jetson-inference #310: CaffeParser: Could not open file      CaffeParser: Could not parse model file

strivezh, 2019-04-14:
./detectnet-console dog_1.jpg output_dog1.jpg coco-dog
detectnet-console
  args (4):  0 [./detectnet-console]  1 [dog_1.jpg]  2 [output_dog1.jpg]  3 [coco-dog]  

detectNet -- loading detection network model from:
          -- prototxt     networks/DetectNet-COCO-Dog/deploy.prototxt
          -- model        networks/DetectNet-COCO-Dog/snapshot_iter_38600.caffemodel
          -- input_blob   'data'
          -- output_cvg   'coverage'
          -- output_bbox  'bboxes'
          -- mean_pixel   0.000000
          -- class_labels networks/DetectNet-COCO-Dog/class_labels.txt
          -- threshold    0.500000
          -- batch_size   2

[TRT]  TensorRT version 5.0.6
[TRT]  detected model format - caffe  (extension '.caffemodel')
[TRT]  desired precision specified for GPU: FASTEST
[TRT]  requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]  native precisions detected for GPU:  FP32, FP16
[TRT]  selecting fastest native precision for GPU:  FP16
[TRT]  attempting to open engine cache file .2.1.GPU.FP16.engine
[TRT]  cache file not found, profiling network model on device GPU
[TRT]  device GPU, loading  
[TRT]  CaffeParser: Could not open file 
[TRT]  CaffeParser: Could not parse model file
[TRT]  device GPU, failed to parse caffe network
device GPU, failed to load 
detectNet -- failed to initialize.
detectnet-console:   failed to initialize detectNet

closed 2019-04-15 by strivezh

matusbielik, 2019-05-06:
I have the very same problem, how did you solve this?

carlos7ags, 2019-05-25:
I have the same problem.