dusty-nv/jetson-inference #312: Issue with nvidia driver on jetson nano

piro-technologies, 2019-04-16:
Getting this error when trying to install drivers to set up NGC on jetson nano...

```
<pre><font color="#8AE234"><b>piro-ai@PIRO-AI</b></font>:<font color="#729FCF"><b>~</b></font>$ sudo apt-get update &amp;&amp; sudo apt-get install -y --no-install-recommends cuda-drivers
Get:1 file:/var/cuda-repo-10-0-local-10.0.166  InRelease
Ign:1 file:/var/cuda-repo-10-0-local-10.0.166  InRelease
Get:2 file:/var/visionworks-repo  InRelease
Ign:2 file:/var/visionworks-repo  InRelease
Get:3 file:/var/visionworks-sfm-repo  InRelease
Ign:3 file:/var/visionworks-sfm-repo  InRelease
Get:4 file:/var/visionworks-tracking-repo  InRelease
Ign:4 file:/var/visionworks-tracking-repo  InRelease
Get:5 file:/var/cuda-repo-10-0-local-10.0.166  Release [574 B]
Get:6 file:/var/visionworks-repo  Release [1,999 B]
Get:7 file:/var/visionworks-sfm-repo  Release [2,003 B]
Get:5 file:/var/cuda-repo-10-0-local-10.0.166  Release [574 B]
Get:8 file:/var/visionworks-tracking-repo  Release [2,008 B]
Get:6 file:/var/visionworks-repo  Release [1,999 B]
Get:7 file:/var/visionworks-sfm-repo  Release [2,003 B]
Get:8 file:/var/visionworks-tracking-repo  Release [2,008 B]
Ign:13 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  InRelease
Get:14 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  Release [564 B]
Get:15 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  Release.gpg [819 B]
Get:16 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  Packages [71.4 kB]
Hit:17 http://ports.ubuntu.com/ubuntu-ports bionic InRelease                                                                                          
Get:18 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease [88.7 kB]                                                                        
Get:19 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease [74.6 kB]                                                                      
Get:20 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease [88.7 kB]                                                                       
Get:21 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main arm64 Packages [468 kB]                                                               
Get:22 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main arm64 DEP-11 Metadata [278 kB]                                                        
Get:23 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main DEP-11 48x48 Icons [66.7 kB]                                                          
Get:24 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main DEP-11 64x64 Icons [123 kB]                                                           
Get:25 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe arm64 Packages [645 kB]                                                           
Get:26 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe arm64 DEP-11 Metadata [209 kB]                                                    
Hit:27 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu bionic InRelease                                                                          
Get:28 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe DEP-11 48x48 Icons [189 kB]                                                       
Get:29 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe DEP-11 64x64 Icons [360 kB]                                                       
Get:30 http://ports.ubuntu.com/ubuntu-ports bionic-backports/universe arm64 DEP-11 Metadata [7,356 B]                                                 
Get:31 http://ports.ubuntu.com/ubuntu-ports bionic-security/main arm64 DEP-11 Metadata [204 B]                                                        
Get:32 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe arm64 DEP-11 Metadata [20.7 kB]                                                  
Get:33 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe DEP-11 48x48 Icons [12.2 kB]                                                     
Get:34 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe DEP-11 64x64 Icons [45.2 kB]                                                     
Fetched 2,750 kB in 18s (156 kB/s)                                                                                                                    
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package cuda-drivers
</pre>
```

dusty-nv, 2019-04-16:
Hi @piro-technologies, NGC / DIGITS isn't supported on DIGITS, it's intended to be installed on an x86_64 host PC.  The host PC should have a discrete NVIDIA GPU attached.

piro-technologies, 2019-04-16:
Hi @dusty-nv , thanks for your reply... You mean "NGC/DIGITS isn't supported on Jetson Nano"? If this is the case then it will not work on any jetson family?

dusty-nv, 2019-04-18:
That's correct, NGC / DIGITS isn't officially supported on Jetson, it is meant to run on a PC, server, or workstation.

closed 2023-04-27 by dusty-nv