dusty-nv/jetson-inference #324: Applying imageNet to a avi file

skprasadu, 2019-05-04:
Hello Dusty,

I have seen several places where they have mentioned that we can use  filesrc to get the frame Capture and pass that to imageNet. I tried this few few days, with no luck.

I also went to gstCamera and modified the buildLaunchStr and put this as below,

"filesrc location=cars.avi ! videoparse width=1920 height=1080 framerate=30/1 format=2 ! appsink name=mysink"

But once I run the imagenet-camera, I get this error "failed to convert from NV12 to RGBA"

Now I am stuck. Can you help me?

Krishna

dusty-nv, 2019-05-04:
Hi Krishna, I think the issue is that you are missing the GStreamer elements in your pipeline to parse the AVI file and decode it.  Check out the Decoding examples from the NVIDIA L4T GStreamer User Guide for examples of that.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

skprasadu, 2019-05-05:
This was a good pointer, it worked, the gstreamer pipeline is as below, just in case if it is useful to others,

"filesrc location=<avi file> ! avidemux name=demux ! decodebin ! videoconvert ! videoscale ! appsink name=mysink"

devshank3, 2019-08-06:
> "filesrc location="avi file" ! avidemux name=demux ! decodebin ! videoconvert ! videoscale ! appsink name=mysink"

Where should we update this code ?
and any ways to modify imagenet-camera to take in internal video file as mp4 as such in the net

closed 2023-04-27 by dusty-nv