dusty-nv/jetson-inference #329: Nothing works... -> ERROR: Check failed: status == CUDNN_STATUS_SUCCESS (8 vs. 0) CUDNN_STATUS_EXECUTION_FAILED, device 0

jldk20, 2019-05-21:
I just follwed the instructions what am i doing wrong? @dusty-nv 

Complete Error:

ERROR: Check failed: status == CUDNN_STATUS_SUCCESS (8 vs. 0) CUDNN_STATUS_EXECUTION_FAILED, device 0

inception_5b/relu_pool_proj <- inception_5b/pool_proj
inception_5b/relu_pool_proj -> inception_5b/pool_proj (in-place)
Setting up inception_5b/relu_pool_proj
TRAIN Top shape for layer 158 'inception_5b/relu_pool_proj' 5 128 40 40 (1024000)
Creating layer 'inception_5b/output' of type 'Concat'
Layer's types are Ftype:FLOAT Btype:FLOAT Fmath:FLOAT Bmath:FLOAT
Created Layer inception_5b/output (159)
inception_5b/output <- inception_5b/1x1
inception_5b/output <- inception_5b/3x3
inception_5b/output <- inception_5b/5x5
inception_5b/output <- inception_5b/pool_proj
inception_5b/output -> inception_5b/output
Setting up inception_5b/output
TRAIN Top shape for layer 159 'inception_5b/output' 5 1024 40 40 (8192000)
Creating layer 'pool5/drop_s1' of type 'Dropout'
Layer's types are Ftype:FLOAT Btype:FLOAT Fmath:FLOAT Bmath:FLOAT
Created Layer pool5/drop_s1 (160)
pool5/drop_s1 <- inception_5b/output
pool5/drop_s1 -> pool5/drop_s1
Check failed: status == CUDNN_STATUS_SUCCESS (8 vs. 0)  CUDNN_STATUS_EXECUTION_FAILED, device 0

EmbeddedDesign, 2019-06-03:
Same issue running under the Nvidia DIGITS Docker container when trying to create a new Object Detection model:
> inception_5b/relu_pool_proj <- inception_5b/pool_proj
> inception_5b/relu_pool_proj -> inception_5b/pool_proj (in-place)
> Setting up inception_5b/relu_pool_proj
> TRAIN Top shape for layer 158 'inception_5b/relu_pool_proj' 2 128 40 40 (409600)
> Creating layer 'inception_5b/output' of type 'Concat'
> Layer's types are Ftype:FLOAT Btype:FLOAT Fmath:FLOAT Bmath:FLOAT
> Created Layer inception_5b/output (159)
> inception_5b/output <- inception_5b/1x1
> inception_5b/output <- inception_5b/3x3
> inception_5b/output <- inception_5b/5x5
> inception_5b/output <- inception_5b/pool_proj
> inception_5b/output -> inception_5b/output
> Setting up inception_5b/output
> TRAIN Top shape for layer 159 'inception_5b/output' 2 1024 40 40 (3276800)
> Creating layer 'pool5/drop_s1' of type 'Dropout'
> Layer's types are Ftype:FLOAT Btype:FLOAT Fmath:FLOAT Bmath:FLOAT
> Created Layer pool5/drop_s1 (160)
> pool5/drop_s1 <- inception_5b/output
> pool5/drop_s1 -> pool5/drop_s1
> Check failed: status == CUDNN_STATUS_SUCCESS (8 vs. 0) CUDNN_STATUS_EXECUTION_FAILED, device 0

Dataset is 640x640 with labels in KITTI format and the data ingestion job runs with no issues.

Model Info:
> Batch Size: **2**
> Batch Accumulation: **5**
> Solver: **Adam**
> Custom Caffe Network: [**detectnet.prototxt**](https://github.com/dusty-nv/jetson-inference/blob/master/data/networks/detectnet.prototxt)
> Pretrained model: [**bvlc_googlenet.caffemodel**
> ](http://dl.caffe.berkeleyvision.org/bvlc_googlenet.caffemodel)
> 

Digits info:
> DIGITS version: **6.1.1**
> Caffe version: **0.17.0**
> Caffe flavor: **NVIDIA**

Output of deviceQuery:
> CUDA Device Query (Runtime API) version (CUDART static linking)
> 
> Detected 1 CUDA Capable device(s)
> 
> Device 0: "GeForce RTX 2070"
>   CUDA Driver Version / Runtime Version          10.1 / 9.0
>   CUDA Capability Major/Minor version number:    7.5
>   Total amount of global memory:                 7944 MBytes (8330215424 bytes)
> MapSMtoCores for SM 7.5 is undefined.  Default to use 64 Cores/SM
> MapSMtoCores for SM 7.5 is undefined.  Default to use 64 Cores/SM
>   (36) Multiprocessors, ( 64) CUDA Cores/MP:     2304 CUDA Cores
>   GPU Max Clock rate:                            1725 MHz (1.73 GHz)
>   Memory Clock rate:                             7001 Mhz
>   Memory Bus Width:                              256-bit
>   L2 Cache Size:                                 4194304 bytes
>   Maximum Texture Dimension Size (x,y,z)         1D=(131072), 2D=(131072, 65536), 3D=(16384, 16384, 16384)
>   Maximum Layered 1D Texture Size, (num) layers  1D=(32768), 2048 layers
>   Maximum Layered 2D Texture Size, (num) layers  2D=(32768, 32768), 2048 layers
>   Total amount of constant memory:               65536 bytes
>   Total amount of shared memory per block:       49152 bytes
>   Total number of registers available per block: 65536
>   Warp size:                                     32
>   Maximum number of threads per multiprocessor:  1024
>   Maximum number of threads per block:           1024
>   Max dimension size of a thread block (x,y,z): (1024, 1024, 64)
>   Max dimension size of a grid size    (x,y,z): (2147483647, 65535, 65535)
>   Maximum memory pitch:                          2147483647 bytes
>   Texture alignment:                             512 bytes
>   Concurrent copy and kernel execution:          Yes with 3 copy engine(s)
>   Run time limit on kernels:                     Yes
>   Integrated GPU sharing Host Memory:            No
>   Support host page-locked memory mapping:       Yes
>   Alignment requirement for Surfaces:            Yes
>   Device has ECC support:                        Disabled
>   Device supports Unified Addressing (UVA):      Yes
>   Supports Cooperative Kernel Launch:            Yes
>   Supports MultiDevice Co-op Kernel Launch:      Yes
>   Device PCI Domain ID / Bus ID / location ID:   0 / 1 / 0
>   Compute Mode:
>      < Default (multiple host threads can use ::cudaSetDevice() with device simultaneously) >
> 
> deviceQuery, CUDA Driver = CUDART, CUDA Driver Version = 10.1, CUDA Runtime Version = 9.0, NumDevs = 1
> Result = PASS

nvidia-smi info:
> NVIDIA-SMI: **418.43**
> Driver Version: **418.43**
> CUDA Version: **10.1**

nvcc --version:
> nvcc: NVIDIA (R) Cuda compiler driver
> Copyright (c) 2005-2017 NVIDIA Corporation
> Built on Fri_Sep__1_21:08:03_CDT_2017
> Cuda compilation tools, release 9.0, V9.0.176

cat /usr/include/cudnn.h | grep CUDNN_MAJOR -A 2:
> #define CUDNN_MAJOR 7
> #define CUDNN_MINOR 1
> #define CUDNN_PATCHLEVEL 2

EmbeddedDesign, 2019-06-04:
Upgraded container to `nvcr.io/nvidia/digits:19.05-caffe` and this issue went away for me.

Jackson-Rong, 2019-08-08:
Hello!  I run the image classification model with the smiliar error message, but the situation seem different.  I also have verified the CUDA & CuDNN installation before running the model.  Listed below are the error log.  Hopefully someone can help me, the newby to Ubuntu & Nvidia AI.

NVIDIA-SMI 418.67       Driver Version: 418.67       CUDA Version: 10.1
cudnnGetVersion() : 7602
Ubuntu 18.04.2 LTS

Creating layer 'loss1/fc' of type 'InnerProduct'
Layer's types are Ftype:FLOAT Btype:FLOAT Fmath:FLOAT Bmath:FLOAT
Created Layer loss1/fc (62)
loss1/fc <- loss1/conv
loss1/fc -> loss1/fc
Setting up loss1/fc
TRAIN Top shape for layer 62 'loss1/fc' 32 1024 (32768)
Creating layer 'loss1/relu_fc' of type 'ReLU'
Layer's types are Ftype:FLOAT Btype:FLOAT Fmath:FLOAT Bmath:FLOAT
Created Layer loss1/relu_fc (63)
loss1/relu_fc <- loss1/fc
loss1/relu_fc -> loss1/fc (in-place)
Setting up loss1/relu_fc
TRAIN Top shape for layer 63 'loss1/relu_fc' 32 1024 (32768)
Creating layer 'loss1/drop_fc' of type 'Dropout'
Layer's types are Ftype:FLOAT Btype:FLOAT Fmath:FLOAT Bmath:FLOAT
Created Layer loss1/drop_fc (64)
loss1/drop_fc <- loss1/fc
loss1/drop_fc -> loss1/fc (in-place)
Check failed: status == CUDNN_STATUS_SUCCESS (8 vs. 0)  CUDNN_STATUS_EXECUTION_FAILED, device 0

closed 2023-04-27 by dusty-nv