dusty-nv/jetson-inference #331: ~/jetson_clocks.sh and Nano?

thovoll, 2019-05-24:
The docs reference ~/jetson_clocks.sh which doesn't seem to be available on Nano. Is that correct? If so, would be good to update the docs.

thovoll, 2019-05-24:
For example here: https://github.com/dusty-nv/jetson-inference/blob/master/docs/detectnet-camera-2.md

dusty-nv, 2019-05-24:
It is now just jetson_clocks
No need for the ~/  or the .sh
Thanks for the heads-up on the docs!

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

thovoll, 2019-05-24:
Ok, thanks! Found it. Is there a way to turn the clocks back down?

dusty-nv, 2019-05-24:
To do that, first you would have needed to run:

jetson_clocks --store

Then run jetson_clocks to maximize, and when you want to turn the clocks back down, run:

jetson_clocks --restore

Assuming you didn’t run it with --store first, reboot and the system will be back to normal.  Then you can run --store

thovoll, 2019-05-24:
Yes, that's what I ended up doing. Thanks!

closed 2023-04-27 by dusty-nv