dusty-nv/jetson-inference #332: How to load ONNX model?

tengdai722, 2019-05-24:
As ONNX format only contains 1 file. Could anyone provide some tutorial? Thank you!!

dusty-nv, 2019-05-25:
ONNX tutorial is still in progress, but essentially you supply ONNX filename in model argument, and ignore the prototxt argument.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

tengdai722, 2019-05-27:
> ONNX tutorial is still in progress, but essentially you supply ONNX filename in model argument, and ignore the prototxt argument.
> […](#)
> ----------------------------------------------------------------------------------- This email message is for the sole use of the intended recipient(s) and may contain confidential information. Any unauthorized review, use, disclosure or distribution is prohibited. If you are not the intended recipient, please contact the sender by reply email and destroy all copies of the original message. -----------------------------------------------------------------------------------

Thank you very much for your reply! I am following the "my-recoginition" file and notice that the only line take the input is "imageNet* net = imageNet::Create();" line. I have read the code and presume that the imageNet::Create() should take two input, one is an integer, and anther should be the file path. In the ONNX case, I passed integer as 2 while it seems won't work. Maybe I don't understand how this is working with taking own .onnx model. Would mind give a little hints? Thank you!

dusty-nv, 2019-06-04:
@tengdai722 see this page for the syntax of running `imagenet-console` with custom image recognition model:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/imagenet-custom.md

dszpak1, 2020-08-12:
Just to be clear you don't have to write the entire filepath, just the filename?

closed 2023-04-27 by dusty-nv