dusty-nv/jetson-inference #334: The Jetson Nano crashes regurlary when running the live camera demo ./detect-camera

vlschmidt, 2019-05-27:
The behaviour is a little bit sporadic, sometimes the demo will run for quite a long time, on other occassions it crashes right after starting. It almost always crashes when I open another window in Ubuntu. What is the reason for these crashes, are there any ways around this. 

I'm sorry if I am posting this in the wrong forum, but I couldn't find a better place as I'm not sure if this is a hardware or a software issue.

dusty-nv, 2019-05-27:
Hi vlschmidt, can you try putting your board in 5W mode and see if that helps?

sudo nvpmodel -m 1

If it doesn't crash in 5W mode, that means it's a power supply issue, and perhaps you can try a 4A DC barrel jack adapter.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

vlschmidt, 2019-05-29:
Thank you! This helped. I did not need a 4A DV barrel adapter. My 5V, 2A usb power adapter worked fine.

closed 2019-05-29 by vlschmidt

Foratog, 2019-12-18:
The  command worked for me with my %V, 2A usb power adapter as well thank you so much...

cydu123, 2020-07-04:
I have the same problem. It's a annoying issue. I'm using 4A DC barrel jack adapter.

dusty-nv, 2020-07-05:
@cydu123 does the issue still happen in 5W mode?  If so, then it is related to power supply's stability, even if it is barrel jack adapter.

Kaju-Bubanja, 2020-10-19:
I also have sporadic crashes changing the power mode from MAXN to 5W did not really help.