dusty-nv/jetson-inference #335: errors when runing segnet: Segmentation fault (core dumped)  and Weights for layer conv1 doesn't exist

vxgu86, 2019-05-30:
two problems for me here

when I run segnet with the pretrained model with :
NET=/home/../jetson-inference/data/networks/FCN-Alexnet-Aerial-FPV-4ch-720p
./segnet-console 0132.png output.png \
--prototxt=$NET/deploy.prototxt \
--model=$NET/snapshot_iter_1777146.caffemodel \
--labels=$NET/fpv-labels.txt \
--colors=$NET/fpv-deploy-colors.txt \
--input_blob=data \
--output_blob=score_fr 

**the error is as followed:**
[TRT]  TensorRT version 5.0.6
[TRT]  detected model format - caffe  (extension '.caffemodel')
[TRT]  desired precision specified for GPU: FASTEST
[TRT]  requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]  native precisions detected for GPU:  FP32, FP16
[TRT]  selecting fastest native precision for GPU:  FP16
[TRT]  attempting to open engine cache file /home/vxgunano/jetson-inference/data/networks/FCN-Alexnet-Aerial-FPV-4ch-720p/snapshot_iter_1777146.caffemodel.2.1.GPU.FP16.engine
[TRT]  cache file not found, profiling network model on device GPU
[TRT]  device GPU, loading /home/vxgunano/jetson-inference/data/networks/FCN-Alexnet-Aerial-FPV-4ch-720p/deploy.prototxt /home/vxgunano/jetson-inference/data/networks/FCN-Alexnet-Aerial-FPV-4ch-720p/snapshot_iter_1777146.caffemodel
[TRT]  failed to retrieve tensor for Output "score_fr"
[TRT]  retrieved Input tensor "data":  3x720x1280
[TRT]  device GPU, configuring CUDA engine
[TRT]  device GPU, building FP16:  ON
[TRT]  device GPU, building INT8:  OFF
[TRT]  device GPU, building CUDA engine (this may take a few minutes the first time a network is loaded)
Segmentation fault (core dumped)

I trained the segnet model as the instruction said, the model worked on tx2, but not nano. 
error:

[TRT]  TensorRT version 5.0.6
[TRT]  detected model format - caffe  (extension '.caffemodel')
[TRT]  desired precision specified for GPU: FASTEST
[TRT]  requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]  native precisions detected for GPU:  FP32, FP16
[TRT]  selecting fastest native precision for GPU:  FP16
[TRT]  attempting to open engine cache file /home/vxgunano/cvexp/exp3_epoch_30.0/snapshot_iter_135660.caffemodel.2.1.GPU.FP16.engine
[TRT]  cache file not found, profiling network model on device GPU
[TRT]  device GPU, loading /home/vxgunano/cvexp/exp3_epoch_30.0/deploy.prototxt /home/vxgunano/cvexp/exp3_epoch_30.0/snapshot_iter_135660.caffemodel
**Weights for layer conv1 doesn't exist
[TRT]  CaffeParser: ERROR: Attempting to access NULL weights
Weights for layer conv1 doesn't exist**
[TRT]  CaffeParser: ERROR: Attempting to access NULL weights
[TRT]  Parameter check failed at: ../builder/Network.cpp::addConvolution::66, condition: kernelWeights.values != nullptr
error parsing layer type Convolution index 1
[TRT]  device GPU, failed to parse caffe network
device GPU, failed to load /home/vxgunano/cvexp/exp3_epoch_30.0/snapshot_iter_135660.caffemodel
segNet -- failed to initialize.
segnet-console:   failed to initialize segnet

dcguru2016, 2019-06-07:
I am having the same issue on the nano. any resolution on this?

dusty-nv, 2019-06-07:
Are you able to run any of the other segmentation models from the repo?

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

nthakor, 2019-06-12:
I am not even able to run segnet-camera. I get `killed` after 
`[TRT] device GPU, building CUDA engine (this may take a few minutes the first time a network is loaded. `

dcguru2016, 2019-06-13:
yes i am able to run other segmentation models

bcwdesign, 2019-06-15:
One thing to confirm is that the clock was set correctly before you did the build. If not the build may complete but not all would have downloaded and built correctly.

youyoubilly, 2019-12-24:
I have the same issue. I have not idea how to solve it....please help...

[TRT]  failed to retrieve tensor for Output "prob"
[TRT]  device GPU, configuring CUDA engine
[TRT]  device GPU, building FP16:  ON
[TRT]  device GPU, building INT8:  OFF
[TRT]  device GPU, building CUDA engine (this may take a few minutes the first time a network is loaded)
Segmentation fault (core dumped)

robelsharma, 2020-09-04:
Anyone have a fix for it?

closed 2023-04-27 by dusty-nv