dusty-nv/jetson-inference #337: bash: ./imagenet-camera: No such file or directory

Santhosh1509, 2019-06-06:
Beginner here, following the tutorial **Running the Live Camera Recognition Demo**

when i run ` ./imagenet-camera googlenet`

I get this line

> bash: ./imagenet-camera: No such file or directory

any help?

dusty-nv, 2019-06-06:
Hi there, is your terminal in <dir-where-installed>/jetson-inference/build/aarch64/bin ?   Did the project compile successfully?  Can you list the output of "ls" command - thanks.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

Santhosh1509, 2019-06-06:
I ran the command outside the directory where the build was made.
It does run inside the build directory
Would be nice if the tutorial would be more specific, as it would be helpful for a newbee like me
Thank you @dusty-nv

closed 2019-06-06 by Santhosh1509