dusty-nv/jetson-inference #338: ./imagenet-camera googlenet GST_ARGUS: Available Sensor modes : Segmentation fault (core dumped)

Santhosh1509, 2019-06-06:
Ran the command

`./imagenet-camera googlenet`
inside

> jetson-inference/build/aarch64/bin

I get this

> imagenet-camera

>   args (2):  0 [./imagenet-camera]  1 [googlenet]  
> 
> [gstreamer] initialized gstreamer, version 1.14.1.0
> [gstreamer] gstCamera attempting to initialize with GST_SOURCE_NVCAMERA
> [gstreamer] gstCamera pipeline string:
> nvcamerasrc fpsRange="30.0 30.0" ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12 ! nvvidconv flip-method=2 ! video/x-raw ! appsink name=mysink
> [gstreamer] gstCamera failed to create pipeline
> [gstreamer]    (no element "nvcamerasrc")
> [gstreamer] failed to init gstCamera (GST_SOURCE_NVCAMERA)
> [gstreamer] gstCamera attempting to initialize with GST_SOURCE_NVARGUS
> [gstreamer] gstCamera pipeline string:
> nvarguscamerasrc ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, framerate=30/1, format=(string)NV12 ! nvvidconv flip-method=2 ! video/x-raw ! appsink name=mysink
> [gstreamer] gstCamera successfully initialized with GST_SOURCE_NVARGUS
> 
> imagenet-camera:  successfully initialized video device
>     width:  1280
>    height:  720
>     depth:  12 (bpp)
> 
> 
> imageNet -- loading classification network model from:
>          -- prototxt     networks/googlenet.prototxt
>          -- model        networks/bvlc_googlenet.caffemodel
>          -- class_labels networks/ilsvrc12_synset_words.txt
>          -- input_blob   'data'
>          -- output_blob  'prob'
>          -- batch_size   2
> 
> [TRT]  TensorRT version 5.0.6
> [TRT]  detected model format - caffe  (extension '.caffemodel')
> [TRT]  desired precision specified for GPU: FASTEST
> [TRT]  requested fasted precision for device GPU without providing valid calibrator, disabling INT8
> [TRT]  native precisions detected for GPU:  FP32, FP16
> [TRT]  selecting fastest native precision for GPU:  FP16
> [TRT]  attempting to open engine cache file networks/bvlc_googlenet.caffemodel.2.1.GPU.FP16.engine
> [TRT]  loading network profile from engine cache... networks/bvlc_googlenet.caffemodel.2.1.GPU.FP16.engine
> [TRT]  device GPU, networks/bvlc_googlenet.caffemodel loaded
> [TRT]  device GPU, CUDA engine context initialized with 2 bindings
> [TRT]  binding -- index   0
>                -- name    'data'
>                -- type    FP32
>                -- in/out  INPUT
>                -- # dims  3
>                -- dim #0  3 (CHANNEL)
>                -- dim #1  224 (SPATIAL)
>                -- dim #2  224 (SPATIAL)
> [TRT]  binding -- index   1
>                -- name    'prob'
>                -- type    FP32
>                -- in/out  OUTPUT
>                -- # dims  3
>                -- dim #0  1000 (CHANNEL)
>                -- dim #1  1 (SPATIAL)
>                -- dim #2  1 (SPATIAL)
> [TRT]  binding to input 0 data  binding index:  0
> [TRT]  binding to input 0 data  dims (b=2 c=3 h=224 w=224) size=1204224
> [cuda]  cudaAllocMapped 1204224 bytes, CPU 0x100e30000 GPU 0x100e30000
> [TRT]  binding to output 0 prob  binding index:  1
> [TRT]  binding to output 0 prob  dims (b=2 c=1000 h=1 w=1) size=8000
> [cuda]  cudaAllocMapped 8000 bytes, CPU 0x100f60000 GPU 0x100f60000
> device GPU, networks/bvlc_googlenet.caffemodel initialized.
> [TRT]  networks/bvlc_googlenet.caffemodel loaded
> imageNet -- loaded 1000 class info entries
> networks/bvlc_googlenet.caffemodel initialized.
> default X screen 0:   1360 x 768
> [OpenGL]  glDisplay display window initialized
> [OpenGL]   creating 1280x720 texture
> loaded image  fontmapA.png  (256 x 512)  2097152 bytes
> [cuda]  cudaAllocMapped 2097152 bytes, CPU 0x101060000 GPU 0x101060000
> [cuda]  cudaAllocMapped 8192 bytes, CPU 0x100f62000 GPU 0x100f62000
> [gstreamer] gstreamer transitioning pipeline to GST_STATE_PLAYING
> [gstreamer] gstreamer changed state from NULL to READY ==> mysink
> [gstreamer] gstreamer changed state from NULL to READY ==> capsfilter2
> [gstreamer] gstreamer changed state from NULL to READY ==> nvvconv1
> [gstreamer] gstreamer changed state from NULL to READY ==> capsfilter1
> [gstreamer] gstreamer changed state from NULL to READY ==> nvarguscamerasrc0
> [gstreamer] gstreamer changed state from NULL to READY ==> pipeline1
> [gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter2
> [gstreamer] gstreamer changed state from READY to PAUSED ==> nvvconv1
> [gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter1
> [gstreamer] gstreamer stream status CREATE ==> src
> [gstreamer] gstreamer changed state from READY to PAUSED ==> nvarguscamerasrc0
> [gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline1
> [gstreamer] gstreamer msg new-clock ==> pipeline1
> [gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter2
> [gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvvconv1
> [gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter1
> [gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvarguscamerasrc0
> [gstreamer] gstreamer stream status ENTER ==> src
> [gstreamer] gstreamer msg stream-start ==> pipeline1
> 
> imagenet-camera:  camera open for streaming
> GST_ARGUS: Creating output stream
> CONSUMER: Waiting until producer is connected...
> GST_ARGUS: Available Sensor modes :
> Segmentation fault (core dumped)

Found a similar issue [Jetson Inference on Xavier Core Dumped (Solved)](https://devtalk.nvidia.com/default/topic/1042156/jetson-agx-xavier/jetson-inference-on-xavier-core-dumped-solved-/)
Tried this [Jetson Xavier + Logitech C920 Issue #267](https://github.com/dusty-nv/jetson-inference/issues/267)

Found linked webcam using `ls -ltrh /dev/video*`
 
changed line inside 

> jetson-inference/imagenet-camera/imagenet-camera.cpp

 `#define DEFAULT_CAMERA -1`
to
`#define DEFAULT_CAMERA dev/video0`

what's going wrong?

dusty-nv, 2019-06-06:
What camera are you using?   Recommend trying the Raspberry Pi Camera Module v2 if possible - it also has better performance than USB.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

Santhosh1509, 2019-06-06:
> Logitech, Inc. Webcam C170

dusty-nv, 2019-06-06:
@Santhosh1509 please see this topic for steps to set the capture resolution to 640x480 for Logitech C170:  

https://devtalk.nvidia.com/default/topic/1052803/jetson-nano/logitech-c170-support-imagenet-camera-failed-to-capture-frame/post/5344276/#5344276

Santhosh1509, 2019-06-06:
Made the changes specified in the link

started building from source again

showing this while `make`

> [  1%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-YV12.cu.o
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> [  2%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaFont.cu.o
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> [  3%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaNormalize.cu.o
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> [  5%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaOverlay.cu.o
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> /home/rlpl123/jetson-inference/utils/cuda/cudaOverlay.cu(47): warning: variable "thick" was declared but never referenced
> 
> /home/rlpl123/jetson-inference/utils/cuda/cudaOverlay.cu(26): warning: function "eq_less" was declared but never referenced
> 
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> /home/rlpl123/jetson-inference/utils/cuda/cudaOverlay.cu(47): warning: variable "thick" was declared but never referenced
> 
> /home/rlpl123/jetson-inference/utils/cuda/cudaOverlay.cu(26): warning: function "eq_less" was declared but never referenced
> 
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> /home/rlpl123/jetson-inference/utils/cuda/cudaOverlay.cu(47): warning: variable "thick" was declared but never referenced
> 
> /home/rlpl123/jetson-inference/utils/cuda/cudaOverlay.cu(26): warning: function "eq_less" was declared but never referenced
> 
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> [  6%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaRGB.cu.o
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> [  7%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaResize.cu.o
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
[... 125 lines cut ...]
> [ 57%] Building CXX object CMakeFiles/jetson-inference.dir/tensorNet.cpp.o
[... cut ...]
> imagenet-camera/CMakeFiles/imagenet-camera.dir/build.make:62: recipe for target 'imagenet-camera/CMakeFiles/imagenet-camera.dir/imagenet-camera.cpp.o' failed
> make[2]: *** [imagenet-camera/CMakeFiles/imagenet-camera.dir/imagenet-camera.cpp.o] Error 1
> CMakeFiles/Makefile2:179: recipe for target 'imagenet-camera/CMakeFiles/imagenet-camera.dir/all' failed
> make[1]: *** [imagenet-camera/CMakeFiles/imagenet-camera.dir/all] Error 2
> Makefile:129: recipe for target 'all' failed
> make: *** [all] Error 2
>

dusty-nv, 2019-06-06:
Do you have a define like this near the top of the imagenet-camera.cpp?

#define DEFAULT_CAMERA 0

Try setting this to 0 or 1 depending on if your USB webcam is /dev/video0 or /dev/video1

Santhosh1509, 2019-06-06:
Giving the same error even after changing

`#define DEFAULT_CAMERA 0` since my webcam is detected as `/dev/video0`

dusty-nv, 2019-06-06:
I would start with a fresh clone of the repo, then apply that 640x480 edit.  Something has gone wrong in imagenet-camera

Santhosh1509, 2019-06-07:
I built it again, getting some warning. Installation done.
Will try it out with other webcam as well.

> [  1%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-YV12.cu.o
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> [  2%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaFont.cu.o
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> [  3%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaNormalize.cu.o
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> [  5%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaOverlay.cu.o
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> /home/rlpl123/jetson-inference/utils/cuda/cudaOverlay.cu(47): warning: variable "thick" was declared but never referenced
> 
> /home/rlpl123/jetson-inference/utils/cuda/cudaOverlay.cu(26): warning: function "eq_less" was declared but never referenced
> 
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> /home/rlpl123/jetson-inference/utils/cuda/cudaOverlay.cu(47): warning: variable "thick" was declared but never referenced
> 
> /home/rlpl123/jetson-inference/utils/cuda/cudaOverlay.cu(26): warning: function "eq_less" was declared but never referenced
> 
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> /home/rlpl123/jetson-inference/utils/cuda/cudaOverlay.cu(47): warning: variable "thick" was declared but never referenced
> 
> /home/rlpl123/jetson-inference/utils/cuda/cudaOverlay.cu(26): warning: function "eq_less" was declared but never referenced
> 
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> [  6%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaRGB.cu.o
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> [  7%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaResize.cu.o
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
> <command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
[... cut ...]
> [ 88%] Linking CXX executable ../../../aarch64/bin/v4l2-display
> [ 89%] Built target v4l2-display
> [ 90%] Linking CXX executable ../../../aarch64/bin/gl-display-test
> [ 92%] Built target gl-display-test
> [ 93%] Linking CXX executable ../aarch64/bin/homography-console
> [ 94%] Built target homography-console
> [ 96%] Linking CXX executable ../aarch64/bin/homography-camera
> [ 97%] Built target homography-camera
> [ 98%] Linking CXX executable ../aarch64/bin/superres-console
> [100%] Built target superres-console
>

dusty-nv, 2019-06-07:
It looks like this built fine without errors.  The warnings aren't a problem.

bcwdesign, 2019-06-15:
Need to change to #define DEFAULT_CAMERA 0

I also had to verify the time was set correctly on the device so the build would happen without errors.

bcwdesign, 2019-06-15:
One other thing....I switched from the micro usb power to the barrell jack since more power may be needed. The rational and steps are outlined here:

https://www.jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/

closed 2019-06-27 by Santhosh1509