dusty-nv/jetson-inference #340: China  can't connect BOX.COM

zouzhe1, 2019-06-18:
Is there another download address?

*************************************************************************************************
正在连接 nvidia.app.box.com (nvidia.app.box.com)|59.24.3.173|:443...

dusty-nv, 2019-06-18:
Can you connect to Google Drive from China?

I use Box for hosting models because Google Drive doesn't allow automated downloading of large files using wget.  User must manually click in browser to download large files from Google Drive.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

zouzhe1, 2019-06-18:
@dusty-nv   yes，i can use google drive  by special method。Aws  s3   ,i can too。

dusty-nv, 2019-06-18:
@zouzhe1 thanks for your reply.  What is the special method you need to use to download from Google Drive?

I would like to find a solution for all users in China that is easier for them to use.  The pan.baidu website does not work for me to access.

dusty-nv, 2019-06-18:
@zouzhe1 are you able to download binaries from GitHub releases?

zouzhe1, 2019-06-18:
@dusty-nv  China   pan baidu  is very good ，but it only China  ， i guess you cant use it。generally，google is not connect  for  all  china user。i think the best method is amazon  AWS  S3 service，if  not us area，cn  S3 can connect，and the speed is very good for the most user of China。

zouzhe1, 2019-06-18:
my special method is use proxy software。。。it  not use as public，so it not good method。 i still support AWS S3.

dusty-nv, 2019-06-18:
Hi @zouzhe1 , unfortunately I don't have the AWS S3 instances available to store them there.

Can you test if you can download this test link without going through special proxy?

https://github.com/pytorch/pytorch/archive/v0.2.0.zip

This link was from a GitHub Releases page for PyTorch.
So if that works for you without needing special proxy, I can use GitHub Releases to host the mirror.

paddygoat, 2019-06-18:
Large data sets should be uploaded as tar.gz files to somewhere like Google Drive. But, normal 'wget' does not work!

export fileid=jhkfdh87983ujbkjjdfkjj
export filename=my_data.tar.gz

wget --save-cookies cookies.txt 'https://docs.google.com/uc?export=download&id='$fileid -O- \
     | sed -rn 's/.*confirm=([0-9A-Za-z_]+).*/\1/p' > confirm.txt

wget --load-cookies cookies.txt -O $filename \
     'https://docs.google.com/uc?export=download&id='$fileid'&confirm='$(<confirm.txt)

dusty-nv, 2019-06-18:
The reason Box.com is used is because it allows for normal wget.  Apparently Google Drive and other filesharing sites require a proxy or VPN from China.

Right now I am putting a mirror up containing the models as a GitHub release page for users that are unable to access their primary location.  Everyone should be able to access GitHub.com.

dusty-nv, 2019-06-18:
Hi guys, see this page for a mirror of the models on GitHub:

https://github.com/dusty-nv/jetson-inference/releases

zouzhe1, 2019-06-19:
@dusty-nv  i forget github,haha 。github is ok。

zouzhe1, 2019-06-19:
@dusty-nv 
https://github.com/dusty-nv/jetson-inference/blob/master/docs/imagenet-training.md#customizing-the-object-classes

https://rawgit.com/dusty-nv/jetson-inference/master/tools/imagenet-download.py

The imagenet-download.py 
It is recommended to use multi-threading to download pictures, because it is too slow to download pictures one by one, now one in a second. Waste computer resources, can be downloaded in one second, such as 20 pic

ZisIsNotZis, 2019-07-03:
Though amazon s3 (which github uses for releases) is **very slow** in China (like 10 kb/s here).

SPRCSY, 2019-07-16:
In fact, uploading the course to bilibili.com may be welcomed.

closed 2023-04-27 by dusty-nv