dusty-nv/jetson-inference #341: trained model

ArtlyStyles, 2019-06-21:
Do we have already trained object detection/localization and segnet model I can use directly on Jetson tx2?

dusty-nv, 2019-06-21:
Yes, see the Hello AI World section of the tutorial.  imageNet, detectNet, and segNet each come with pre-trained models to use.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

ArtlyStyles, 2019-06-21:
DO you have tensorflow model?

dusty-nv, 2019-06-25:
Yes for detectNet, there are `ssd-mobilenet-v1`, `ssd-mobilenet-v2`, and `ssd-inception-v2` models.  They are currently found in the python dev branch:

`git clone -b python https://github.com/dusty-nv/jetson-inference`

closed 2023-04-27 by dusty-nv