dusty-nv/jetson-inference #342: ImportError: No module named jetson

Santhosh1509, 2019-06-21:
Followed 
[Building the Project from Source](https://github.com/dusty-nv/jetson-inference/blob/python/docs/building-repo-2.md)

Did

>$ sudo apt-get install git cmake
> $ git clone https://github.com/dusty-nv/jetson-inference
> $ cd jetson-inference
> $ git submodule update --init

guess i forgot to do this in the first time during the build
> $ sudo apt-get install libpython3-dev python3-numpy

> $ cd jetson-inference    # omit if pwd is already jetson-inference from above
> $ mkdir build
> $ cd build
> $ cmake ../

> $ make
> $ sudo make install

I can't find `jetson.inference` and `jetson.utils` inside `/usr/lib/python3.6/dist-packages`

This is what i can see inside after running `ls /usr/lib/python3.6/dist-packages`

> graphsurgeon                  tensorrt                    uff
> graphsurgeon-0.3.2.dist-info  tensorrt-5.0.6.3.dist-info  uff-0.5.5.dist-info

tried to deleting `rm -rf jetson-inference` reinstall with all the following steps
Yet No `jetson.inference` and `jetson.utils` inside `/usr/lib/python3.6/dist-packages`
What is going wrong? Have i missed any step in between?

dusty-nv, 2019-06-21:
Is the Python 2.7 version working?

If you forgot to install libpython3-dev initially, re-run "cmake ../" and "sudo make install"

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

Santhosh1509, 2019-06-25:
I did as you suggested `cmake ../` and `sudo make install` inside new `build`

Content of the dist-packages of all the python

    ls /usr/lib/python2.7/dist-packages

> asn1crypto                                numpy-1.13.3.egg-info
> asn1crypto-0.24.0.egg-info                pip
> attr                                      pip-9.0.1.egg-info
> attrs-17.4.0.egg-info                     pkg_resources
> cairo                                     pluggy
> _cffi_backend.aarch64-linux-gnu.so        pluggy-0.6.0.egg-info
> Crypto                                    py
> cryptography                              py-1.5.2.egg-info
> cryptography-2.1.4.egg-info               pycairo-1.16.2.egg-info
> cv2.aarch64-linux-gnu.so                  pycrypto-2.6.1.egg-info
> dbus                                      pygobject-3.26.1.egg-info
> _dbus_bindings.aarch64-linux-gnu.so       pygtkcompat
> _dbus_glib_bindings.aarch64-linux-gnu.so  pygtk.pth
> easy_install.py                           pygtk.py
> easy_install.pyc                          pygtk.pyc
> enum                                      _pytest
> enum34-1.1.6.egg-info                     pytest-3.3.2.egg-info
> funcsigs                                  pytest.py
> funcsigs-1.0.2.egg-info                   pytest.pyc
> gi                                        pyxdg-0.25.egg-info
> glib                                      README
> gobject                                   secretstorage
> gps                                       SecretStorage-2.3.1.egg-info
> gps-3.17.egg-info                         setuptools
> graphsurgeon                              setuptools-39.0.1.egg-info
> graphsurgeon-0.3.2.dist-info              six-1.11.0.egg-info
> gtk-2.0                                   six.py
> idna                                      six.pyc
> idna-2.6.egg-info                         talloc.aarch64-linux-gnu.so
> ipaddress-1.0.17.egg-info                 tensorrt
> ipaddress.py                              tensorrt-5.0.6.3.dist-info
> ipaddress.pyc                             uff
> keyring                                   uff-0.5.5.dist-info
> keyring-10.6.0.egg-info                   unity_lens_photos-1.0.egg-info
> keyrings                                  wheel
> keyrings.alt-3.0.egg-info                 wheel-0.30.0.egg-info
> lsb_release.py                            xdg
> numpy

    ls /usr/lib/python3.6/dist-packages

> graphsurgeon                  tensorrt                    uff
> graphsurgeon-0.3.2.dist-info  tensorrt-5.0.6.3.dist-info  uff-0.5.5.dist-info

    ls /usr/lib/python3/dist-packages

> apport
> apport_python_hook.py
> apt
> aptdaemon
> apt_inst.cpython-36m-aarch64-linux-gnu.so
> apt_inst.pyi
> apt_pkg.cpython-36m-aarch64-linux-gnu.so
> apt_pkg.pyi
> aptsources
> AptUrl
> apturl-0.5.2.egg-info
> asn1crypto
> asn1crypto-0.24.0.egg-info
> beautifulsoup4-4.6.0.egg-info
> blinker
> blinker-1.4.egg-info
> Brlapi-0.6.6.egg-info
> brlapi.cpython-36m-aarch64-linux-gnu.so
[... 136 lines cut ...]
> unity_scope_yelp-0.1.egg-info
> unity_scope_zotero-0.1.egg-info
> unohelper.py
> uno.py
> UpdateManager
> urllib3
> urllib3-1.22.egg-info
> wadllib
> wadllib-1.3.2.egg-info
> webencodings
> webencodings-0.5.egg-info
> wheel
> wheel-0.30.0.egg-info
> xdg
> xkit
> xkit-0.0.0.egg-info
> yaml
> _yaml.cpython-36m-aarch64-linux-gnu.so
> zope
> zope.interface-4.3.2.egg-info
> zope.interface-4.3.2-nspkg.pth

No `jetson` anywhere.

What's going wrong?

dusty-nv, 2019-06-25:
From your original message, it doesn't look like you cloned the `python` branch.  It is still in development, so for now clone it like this:

`$ git clone -b python https://github.com/dusty-nv/jetson-inference`

Santhosh1509, 2019-06-26:
@dusty-nv Thank you

    find / -type d -name "jetson" -print 2>/dev/null

> /usr/lib/python2.7/dist-packages/jetson
> /usr/lib/python3.6/dist-packages/jetson

I am new to this, it would be convenient for others like me if you could edit [Building the Project from Source](https://github.com/dusty-nv/jetson-inference/blob/python/docs/building-repo-2.md) for python.

closed 2019-06-27 by Santhosh1509

csmanu, 2020-01-28:
When I run './my-recognition.py polar_bear.jpg' as explained under the "Coding Your Own Image Recognition Program (Python)" section, I get the following error:
`./my-recognition.py polar_bear.jpg
Traceback (most recent call last):
  File "./my-recognition.py", line 3, in <module>
    import jetson.inference
ModuleNotFoundError: No module named 'jetson'`

Python version: 3.6.9
OS: Ubuntu 18.04 LTS
Platform: NVIDIA Jetson Nano

As suggested in [](https://devtalk.nvidia.com/default/topic/1068718/jetson-nano/error-when-trying-to-run-hello-ai-world/) , I tried re-installing jetson-inference. I also have checked for the files "jetson.inference" and "jetson.utils" and they exist under "/usr/lib/python3.6/dist-packages/jetson".

Appreciate help. Thanks in advance.

dusty-nv, 2020-01-28:
Can you try running it as:  python3 my-recognition.py

If you run an interactive python3 interpreter, are you able to import jetson.inference ?

csmanu, 2020-01-29:
When I run,
$ python3 my-recognition.py polar_bear.jpg

I get the following error:
Traceback (most recent call last):
  File "./my-recognition.py", line 3, in <module>
    import jetson.inference
ModuleNotFoundError: No module named 'jetson'

dusty-nv, 2020-01-29:
Hmm.  If you run an interactive python3 interpreter/shell, are you able to import jetson.inference ?

Are there any files under your `/usr/lib/python3.6/dist-packages/jetson` ?

csmanu, 2020-01-29:
> Hmm. If you run an interactive python3 interpreter/shell, are you able to import jetson.inference ?
> 
> Are there any files under your `/usr/lib/python3.6/dist-packages/jetson` ?

`Python 3.6.9 (default, Nov  7 2019, 10:44:02) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import jetson.inference
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'jetson' `

I find the following files listed under `/usr/lib/python3.6/dist-packages/jetson`
inference  
__init__.py  
utils

moresandeep, 2020-02-16:
@dusty-nv and @csmanu I ran into this issue today. It turns out I have multiple python3 installations the latest one being 3.8. So, my python3 references to 3.8. All the libs for jetson are installed under 3.6 (don't know why) but when I run the program using python3.6 (as opposed to just python3) I no longer get this error.

federicolucca, 2020-02-17:
Same problem. But i haven't any folder `/jetson` in any `/dist-packages`. i have under python3.6 a folder with `Jetson` ,with uppercase, with a subfolder `GPIO`

silent-code, 2020-07-21:
If using a virtual environment, you have to copy these files after the sudo ldconfig:

cp  /usr/lib/python3.6/dist-packages/jetson_inference_python.so /home/.virtualenvs/dl4cv/lib/python3.6/site-packages/jetson_inference_python.so

cp  /usr/lib/python3.6/dist-packages/jetson_utils_python.so     /home/.virtualenvs/dl4cv/lib/python3.6/site-packages/jetson_utils_python.so

CalebBates, 2021-06-18:
Hello! I have Ubuntu 18.04 LTS and I am using a Jetson TX2

I am getting the following error:
ModuleNotFoundError: No module named 'Jetson'

I run the following command:
sudo pip3 install Jetson

I get the following  errors:
1. Could not find a version that satisfies the requirement Jetson (from versions: none)
2. No matching distribution found for Jetson

What could be possible solutions for fixing the No module named 'Jetson' error?

Thanks for the help!

CalebBates, 2021-07-27:
I am getting the following error:

ModuleNotFoundError: No module named 'Jetson'

How can I fix this?

Thanks!

maximofn, 2022-03-25:
> From your original message, it doesn't look like you cloned the `python` branch. It is still in development, so for now clone it like this:
> 
> `$ git clone -b python https://github.com/dusty-nv/jetson-inference`

Doing this and recompiling it fixed it for me

maximofn, 2022-03-29:
Hi @dusty-nv, using the master branch I had the same problem, I had to download the python branch

sprakashdash, 2022-04-22:
@dusty-nv Is there a way to link the jetson inference module to the latest python. I am using python 3.8

dusty-nv, 2022-04-22:
> @dusty-nv Is there a way to link the jetson inference module to the latest python. I am using python 3.8

Hi @sprakashdash, add `3.8` to these lines:

* https://github.com/dusty-nv/jetson-inference/blob/6cf3b12f503d64903ca1e77d4c5474d9b1513a4c/python/CMakeLists.txt#L11
* https://github.com/dusty-nv/jetson-utils/blob/def4a04d023960781a44f9cd97fd1464093becf0/python/CMakeLists.txt#L11

Also make sure you have the dev libraries installed for python3.8 and numpy for 3.8

Laowai01, 2022-05-20:
I had this issue and the problem was build the jetson inference repository without adding the path of cuda in .bashrc. adding path and doing cmake again solved the issue.

CornelisKuijpers, 2023-02-08:
I had the same issue, issue was my mounted drive was in incorrect format I had to format it to EXT4 then the build was correct