dusty-nv/jetson-inference #343: How to download segnet pre-trained model

ArtlyStyles, 2019-06-25:
In the "Hello AI tutorial" segnet, page, it said "In addition to the aerial model from this tutorial, the repo also includes pre-trained models on other segmentation datasets, including Cityscapes, SYNTHIA, and Pascal-VOC.". But I am not sure how to download pre-trained "Cityscapes" model. Can anyone help?

dusty-nv, 2019-06-25:
See the python branch of the repo.  It has a tool which will download it for you.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

ArtlyStyles, 2019-06-25:
Sorry, I could not find it. Could you point me the file to download the trained model with Cityscapes data?

dusty-nv, 2019-06-25:
To clone the python branch:  `git clone -b python https://github.com/dusty-nv/jetson-inference`

To manually download the Cityscapes models, see [`https://github.com/dusty-nv/jetson-inference/releases`](https://github.com/dusty-nv/jetson-inference/releases)

closed 2023-04-27 by dusty-nv