dusty-nv/jetson-inference #344: compatibility with OpenCV V.4

silvermind20, 2019-06-25:
How can I change the cmake to accept opencv version 4?

dusty-nv, 2019-06-25:
Hi @silvermind20 , try changing `3.0.0` to `4.0.0` in [`CMakeLists.txt:72`](https://github.com/dusty-nv/jetson-inference/blob/bda3d60a6967d5c081162a940f0bbca399081a55/CMakeLists.txt#L66)

silvermind20, 2019-06-25:
Thanks. I changed the 2 CMakeLists.txt files manually to 4.0.0 and it worked fine. But is there anyway to do this in command line when running for example ./imagenet instead of manually change the CMakeLists files?

dusty-nv, 2019-06-25:
This is a build-time change, so needs to be done in the CMakeList file before compiling the imagenet programs.

dusty-nv, 2019-06-25:
BTW it is intended that the CMake `find_package()` would use any OpenCV >3.0.0, but unfortunately it detects 4.0.0 as an incompatible version for some reason.

silvermind20, 2019-06-25:
Thanks a lot for your prompt replies.

closed 2019-06-25 by silvermind20

vihari1729, 2019-11-19:
> Hi @silvermind20 , try changing `3.0.0` to `4.0.0` in [`CMakeLists.txt:72`](https://github.com/dusty-nv/jetson-inference/blob/bda3d60a6967d5c081162a940f0bbca399081a55/CMakeLists.txt#L66)

after changing 3.0.0 to 4.0.0 
my error is :
Could not find a configuration file for package "OpenCV" that is compatible
  with requested version "4.0.0".

  The following configuration files were considered but not accepted:

    /usr/share/OpenCV/OpenCVConfig.cmake, version: 2.4.9.1

-- Configuring incomplete, errors occurred!
See also "/home/unizen/jetson-inference/build/CMakeFiles/CMakeOutput.log".
See also "/home/unizen/jetson-inference/build/CMakeFiles/CMakeError.log".

could you please help me in this

Dwaynekj, 2020-02-10:
@vihari1729 Resurrecting this thread to point out there is a second CMakeLists file located in `tools/trt-console/CMakeLists.txt` that needs to get edited along with the one in the project root