dusty-nv/jetson-inference #346: interference existing video file

Santhosh1509, 2019-06-26:
I want to test inference of an already captured video.

[detectnet-camera.py](https://github.com/dusty-nv/jetson-inference/blob/python/python/examples/detectnet-camera.py)

img here is a `<class 'PyCapsule'>`, 
[img, width, height = camera.CaptureRGBA()](https://github.com/dusty-nv/jetson-inference/blob/python/python/examples/detectnet-camera.py#L52)

I am using opencv to play the video.
where `img` is a `numpy.ndarray`

[PyCapsule](https://docs.python.org/3.1/c-api/capsule.html) is a subtype of PyObject represents an opaque value, useful for C extension modules who need to pass an opaque value (as a void* pointer) through Python code to other C code. 

How do i convert `numpy.ndarray' to `PyCapsule`?

dusty-nv, 2019-06-26:
See cuda-from-numpy.py script, found in aarch64/bin or utils/python/examples

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

Santhosh1509, 2019-06-27:
@dusty-nv Thank You

closed 2019-06-27 by Santhosh1509

MaulyGabriel, 2020-05-11:
Example:

```
from imutils.video import VideoStream
import imutils
import time
import cv2

import jetson.inference
import jetson.utils
import argparse
import sys

parser = argparse.ArgumentParser(description="Classify a live camera stream using an image recognition DNN.", 
						   formatter_class=argparse.RawTextHelpFormatter, epilog=jetson.inference.imageNet.Usage())

parser.add_argument("--network", type=str, default="googlenet", help="pre-trained model to load (see below for options)")
parser.add_argument("--camera", type=str, default="0", help="index of the MIPI CSI camera to use (e.g. CSI camera 0)\nor for VL42 cameras, the /dev/video device to use.\nby default, MIPI CSI camera 0 will be used.")

try:
	opt = parser.parse_known_args()[0]
except:
	print("")
	parser.print_help()
	sys.exit(0)

net = jetson.inference.imageNet(opt.network, sys.argv)

video = 'my_video.mp4'

print("[INFO] starting video stream...")
vs = VideoStream(src=video).start()

time.sleep(2.0)

while True:
	frame = vs.read()

	out_frame = frame.copy()
	out_frame = imutils.resize(out_frame, width=480)

	frame = cv2.cvtColor(frame, cv2.COLOR_BGR2RGB)
	frame = cv2.cvtColor(frame, cv2.COLOR_RGB2RGBA)
	
	cuda_frame = jetson.utils.cudaFromNumpy(frame)

	class_idx, confidence = net.Classify(cuda_frame, 1280, 720)

	class_desc = net.GetClassDesc(class_idx)         
	
	cv2.putText(out_frame, "{:05.2f}% {:s}".format(confidence * 100, class_desc), (15, 30), 
        cv2.FONT_HERSHEY_SIMPLEX, 0.6, (0, 255, 0), 1)
	cv2.imshow("Frame", out_frame)
	key = cv2.waitKey(1) & 0xFF

	if key == ord("q"):
		break
vs.stop()
cv2.destroyAllWindows()

```