dusty-nv/jetson-inference #347: Can segnet-console run on jetson nano with Jetpack4.1?

Alphonses, 2019-06-26:
scht@scht-desktop:~/jetson-inference/build/aarch64/bin$ NET=all_water_data
scht@scht-desktop:~/jetson-inference/build/aarch64/bin$ ./segnet-console 11.png output_11.png --prototxt=$NET/deploy.prototxt  --model=$NET/water.caffemodel --labels=$NET/label_names.txt --colors=$NET/color_map.txt --input_blob=data --output_blob=score_f 
segnet-console
  args (9):  0 [./segnet-console]  1 [11.png]  2 [output_11.png]  3 [--prototxt=all_water_data/deploy.prototxt]  4 [--model=all_water_data/water.caffemodel]  5 [--labels=all_water_data/label_names.txt]  6 [--colors=all_water_data/color_map.txt]  7 [--input_blob=data]  8 [--output_blob=score_f]  

segNet -- loading segmentation network model from:
       -- prototxt:   all_water_data/deploy.prototxt
       -- model:      all_water_data/water.caffemodel
       -- labels:     all_water_data/label_names.txt
       -- colors:     all_water_data/color_map.txt
       -- input_blob  'data'
       -- output_blob 'score_f'
       -- batch_size  2

[TRT]  TensorRT version 5.0.6
[TRT]  detected model format - caffe  (extension '.caffemodel')
[TRT]  desired precision specified for GPU: FASTEST
[TRT]  requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]  native precisions detected for GPU:  FP32, FP16
[TRT]  selecting fastest native precision for GPU:  FP16
[TRT]  attempting to open engine cache file all_water_data/water.caffemodel.2.1.GPU.FP16.engine
[TRT]  cache file not found, profiling network model on device GPU
[TRT]  device GPU, loading all_water_data/deploy.prototxt all_water_data/water.caffemodel
[TRT]  CaffeParser: Could not parse binary model file
[TRT]  CaffeParser: Could not parse model file
[TRT]  device GPU, failed to parse caffe network
device GPU, failed to load all_water_data/water.caffemodel
segNet -- failed to initialize.
segnet-console:   failed to initialize segnet
scht@scht-desktop:~/jetson-inference/build/aarch64/bin$ 
________________________________________________________________
Any  help or  tips  is grately  appreciate!  Thanks~

Alphonses, 2019-06-26:
@dusty-nv

dusty-nv, 2019-06-26:
For some reason it appears unable to parse your caffemodel.  How was it trained, and with which version of libprotobuf?

You can also try to see if trt-exec (/usr/src/tensorrt/bin) is able to load the model OK.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

Alphonses, 2019-06-27:
@dusty-nv  Thank your reply in time,I trained my  caffe model by DIGITS 6.1.1 (your tutorial(train+inference)),and my version of libprotobuf is 3.7.1.
 what's more , I  want to know,  this repo do not support jetson nano with jetpack4.1?
But  this repo do support jetson nano with jetpack4.2. So,what's the reason?

dusty-nv, 2019-06-27:
> But this repo do support jetson nano with jetpack4.2. So,what's the reason?
The reason is that JetPack 4.2 is the first version of JetPack to support Jetson Nano.

Alphonses, 2019-06-27:
Thank you sincerely,I get it.

closed 2023-04-27 by dusty-nv