dusty-nv/jetson-inference #350: runtime.cpp (30) - Cuda Error in free: 4 terminate called after throwing an instance of 'nvinfer1::CudaError'

Santhosh1509, 2019-06-27:
[detectnet-camera.py](https://github.com/dusty-nv/jetson-inference/blob/python/python/examples/detectnet-camera.py)

Works fine

When i change my code to run the inference on a recorded video

```
cap = cv2.VideoCapture('output133001.avi')

# process frames until user exits
while display.IsOpen():
	while(True):
		# Capture frame-by-frame
		ret, frame = cap.read()
        
		if ret:
			width,height = frame.shape[:2]
			img = jetson.utils.cudaFromNumpy(frame)
			
			# capture the image
			# ~ img, width, height = camera.CaptureRGBA()
			# ~ print(type(img)," ",type(width)," ",type(height))
			# ~ import sys
			# ~ sys.exit()
			# detect objects in the image (with overlay)
			detections = net.Detect(img, width, height)

			# print the detections
			print("detected {:d} objects in image".format(len(detections)))

			for detection in detections:
				print(detection)

			# render the image
			display.RenderOnce(img, width, height)

			# update the title bar
			display.SetTitle("{:s} | Network {:.0f} FPS".format(opt.network, 1000.0 / net.GetNetworkTime()))

			# print out performance info
			net.PrintProfilerTimes()

			if cv2.waitKey(1) & 0xFF == ord('q'):
				break
```
I get this error with almost all the object detection models

> [OpenGL] glDisplay -- X screen 0 resolution:  1600x900
> [OpenGL] glDisplay -- display device initialized
> jetson.utils -- cudaFromNumpy()  ndarray dim 0 = 480
> jetson.utils -- cudaFromNumpy()  ndarray dim 1 = 640
> jetson.utils -- cudaFromNumpy()  ndarray dim 2 = 3
> [TRT]   engine.cpp (555) - Cuda Error in execute: 4
> [TRT]   engine.cpp (555) - Cuda Error in execute: 4
> [TRT]   detectNet::Detect() -- failed to execute TensorRT context
> Traceback (most recent call last):
>   File "./detectnet-camera.py", line 82, in <module>
>     detections = net.Detect(img, width, height)
> Exception: jetson.inference -- detectNet.Detect() encountered an error classifying the image
> PyTensorNet_Dealloc()
> [cuda]  cudaFreeHost(mDetectionSets[0])
> [cuda]     unspecified launch failure (error 4) (hex 0x04)
> [cuda]     /home/rlpl123/jetson-inference/detectNet.cpp:66
> [cuda]  cudaFreeHost(mClassColors[0])
> [cuda]     unspecified launch failure (error 4) (hex 0x04)
> [cuda]     /home/rlpl123/jetson-inference/detectNet.cpp:74
> [TRT]   runtime.cpp (30) - Cuda Error in free: 4
> terminate called after throwing an instance of 'nvinfer1::CudaError'
>   what():  std::exception
> Aborted

Type of data being fed into the detection using camera and the recorded video is the same.
`img` -> `PyCapsule`
'height` -> `int`
`width` -> `int`

any solution?

Santhosh1509, 2019-06-27:
Though the input was actually expecting 4 channels in the image, I tried to add alpha channel.

`frame = cv2.cvtColor(frame, cv2.COLOR_BGR2RGBA)`

Input image size are `640 x 480`

A frame from the output video

[image]

Do not know what to do!

Santhosh1509, 2019-06-27:
My bad

Changed 
`width,height = frame.shape[:2]`
to
`height,width = frame.shape[:2]`

It started working Fine
Also adding the alpha channel
`frame = cv2.cvtColor(frame, cv2.COLOR_BGR2RGBA)`

closed 2019-06-27 by Santhosh1509

devshank3, 2019-08-06:
which detectnet.py code did you change and run ???

the one inside python/examples or aarch64/bin/ one???