dusty-nv/jetson-inference #353: Low Accuracy Level of Training Custom Model using Custom Dataset

spurani, 2019-07-07:
I am following this guide https://github.com/dusty-nv/jetson-inference/blob/master/docs/imagenet-training.md. I have attached pictures of my custom trained model and custom dataset. I used Caffe Framework and standarad neural Network GoogleNET. I set epoch to 30 and validation data to 10%.

So basically I am trying to identify the type of bottle based on image classification where I have 2 directories (BEER and POP) under one has set of images of BEER bottles and Second one has set of images of POP or SODA bottles. I downloaded these images from IMAGENET. I have put the prediction results of both the type of bottles below. The predictions are made from completely different test images( source google) which I fed after my model got trained. So my question is why the models predictions are not accurate as per the amount of data it has been trained on and 10% validated data? Does the amount of validation dataset and training dataset matters while training the model? or I should try to classify the (for example, instead of detecting 122 individual types of bottles, combining them all into one common bottle class).

[image]

[image]
  
[image]

[image]

dusty-nv, 2019-07-08:
Hi spurani, it looks like you have 1700 training images, which may be low to support 122 different types of bottles. I would try combining them into fewer classes and/or adding more training data.

If you test on some images from your validation set (the network wasn't actually trained on val), does it predict better?  This would indicate the training/val images look considerably different than the test images that you downloaded separately.

[Pop]<https://user-images.githubusercontent.com/28580888/60772369-60866c80-a0c3-11e9-9c69-11a50343d29d.png>

[Beer]<https://user-images.githubusercontent.com/28580888/60772370-62e8c680-a0c3-11e9-9502-6907126b1c8f.png>

[Screenshot from 2019-07-07 13-49-04]<https://user-images.githubusercontent.com/28580888/60772193-f7055e80-a0c0-11e9-8319-214d9957015c.png>

[Dataset]<https://user-images.githubusercontent.com/28580888/60772195-f8368b80-a0c0-11e9-9c07-7744b249b2b8.png>

-----------------------------------------------------------------------------------
This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

spurani, 2019-07-09:
@dusty-nv thanks for your quick response. So based on your suggestion I narrowed down the bottle categories to 2 types(Beer and Pop) which has 1700 training images combined and I tested some of the images from my validation dataset which has 189 images. Still my trained model is not able to classify the images accurately and its making completely opposite predictions which is quite uncertain. As you mentioned earlier that network doesn't actually gets trained on validation dataset so does validation dataset gets used at some point of time where network is actually getting trained on the training dataset and learning to make predictions?

dusty-nv, 2019-07-09:
The validation images are only used during training to evaluate performance of the training, and the model isn’t directly trained on the val images.

It sounds like the class categories might be reversed?   Maybe they are listed in opposite order in your class label file?  If you try switching the labels, do the results make sense?

spurani, 2019-07-12:
> […](#)
> ----------------------------------------------------------------------------------- This email message is for the sole use of the intended recipient(s) and may contain confidential [...]
I checked my directory from where I am uploading the images Dataset it seems to have appropriate labeling. Basically I am downloading only the specific directory of images(Beer(n02823428) , pop(n03983396)) using synset ID of the class directly from ilsvrc12_urls.txt by running the commands from your jetson-inference guide and renaming it accordingly by putting these 2 directories under Bottle directly. I have put the image which you provided in your guide and highlighted training image directory but for me I am uploading from locally I believe that wouldn't make any difference. May I know where can I find class label file.

[image]

spurani, 2019-07-12:
Well I redownloaded the dataset following your jetson-inference guide and created new model retrained the model. I don't have the same issue where it was not able to classify validation dataset images or showing inaccurate classification or reverse classification. I also checked the directory path and directory names which seems appropriate. But now the issue I observed is For example If I pass any image with from validation dataset having extreme noise in the background or having surrounded by multiple random objects or same type of object then my network finds it bit difficult to recognize original target so in results its either showing 50% pop 50% beer or completely reverse or inaccurate. But this situation is kind of unpredictable as for 2 to 3 images it showed those inaccurate results. While testing model on completely new dataset still has lags alot by giving inaccurate predictions even though images which are been fed are completely fresh without any noise in the background or obstacles. So may I know if I am on the right track of learning or if my practices are right? thanks @dusty-nv

closed 2019-07-21 by spurani