dusty-nv/jetson-inference #354: Inference from a different layers - feature extraction

Source82, 2019-07-08:
Hi, 
Please can one edit the imagenet classification code to output features from intermediate layers.

Thanks

Source82, 2019-07-09:
solved by myself. Thanks

closed 2019-07-09 by Source82

deaffella, 2020-08-04:
> solved by myself. Thanks

hi! I have encountered the same problem and can't find a solution.
could you show me your solution to extract features from the intermediate layers?

Sycarol, 2021-01-06:
Hi I have the same problem too, will you please share your solution?

dusty-nv, 2021-01-06:
It would involve editing jetson-inference/c/imageNet.cpp to specify more output layers - like here:

https://github.com/dusty-nv/jetson-inference/blob/0fd4a76101be5c9af4723a9e8376018dfa462be4/c/imageNet.cpp#L138

There is another overload of tensorNet::LoadNetwork() to use that accepts an array of output layer names:

https://github.com/dusty-nv/jetson-inference/blob/0fd4a76101be5c9af4723a9e8376018dfa462be4/c/tensorNet.h#L237