dusty-nv/jetson-inference #355: Failed to build

lhuang-pvamu, 2019-07-10:
I tried to build it following the instruction, but it failed in the following steps. The jetson-utils was not installed so that I have to manually install it. After it, the timeDiff() function doesn't agree with 3 parameters used in tensorNet.h. Please help, thanks!
============

nano:~/jetson-inference/build$ make
[ 38%] Built target jetson-utils
[ 39%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_detectNet.cu.o
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
/home/lhuang/jetson-inference/c/tensorNet.h(453): error: too many arguments in function call

/home/lhuang/jetson-inference/c/tensorNet.h(454): error: identifier "timeFloat" is undefined

2 errors detected in the compilation of "/tmp/tmpxft_00006a03_00000000-8_detectNet.compute_72.cpp1.ii".
CMake Error at jetson-inference_generated_detectNet.cu.o.cmake:279 (message):
  Error generating file
  /home/lhuang/jetson-inference/build/CMakeFiles/jetson-inference.dir/c/./jetson-inference_generated_detectNet.cu.o

CMakeFiles/jetson-inference.dir/build.make:63: recipe for target 'CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_detectNet.cu.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_detectNet.cu.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

agtbaskara, 2019-07-10:
I got exactly the same error on Jetson Nano

dusty-nv, 2019-07-10:
Had you run the "git submodule update --init" step?

Those ISO C++11 whitespace warnings, it tells me an old version of jetson utils is being used.

Can you try recloning jetson-inference repo with this command:

"git clone --recursive https://github.com/dusty-nv/jetson-inference"

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

dusty-nv, 2019-07-10:
Somehow the link to the `jetson-utils` submodule didn't get updated when I merged the changes yesterday, sorry about that.  I have now updated it manually, so please re-clone the `jetson-inference` repo and try building again:

`git clone --recursive https://github.com/dusty-nv/jetson-inference`

lhuang-pvamu, 2019-07-10:
Works perfectly now. Thanks!

Regards,
Lei Huang

closed 2019-07-18 by dusty-nv