dusty-nv/jetson-inference #356: jetson.utils.cudaToNumpy not working as expected

Santhosh1509, 2019-07-10:
[https://github.com/dusty-nv/jetson-utils/blob/master/python/examples/cuda-to-numpy.py#LC44](url)

    array = jetson.utils.cudaToNumpy(cuda_mem, opt.width, opt.height, opt.depth)

Try to save the image giving black image, so tried to plot the array getting converted using matplotlib

    plt.imshow(array)

It gave me a blank image with this in the terminal

> Clipping input data to the valid range for imshow with RGB data ([0..1] for floats or [0..255] for integers).

Saw a solution [here](https://stackoverflow.com/questions/49643907/clipping-input-data-to-the-valid-range-for-imshow-with-rgb-data-0-1-for-floa#answer-51255361)

    plt.imshow((array * 255).astype(np.uint8))

This is the image i am able to get, which is wrong

[image]

This is the image getting saved
[image]

What is going wrong?

[Posted here also](https://github.com/dusty-nv/jetson-utils/issues/7)

Santhosh1509, 2019-07-10:
My bad 

    array = jetson.utils.cudaToNumpy(img, width, height,4)

depth is always 4

closed 2019-07-10 by Santhosh1509

erivera1802, 2019-07-19:
Hi, 
could you tell me, what is img in your case?
I try to convert the output of camera.CaptureRGBA but I got an error. 
```
img, width, height = camera.CaptureRGBA()
numpyImg  = jetson.utils.cudaToNumpy(img,width,height,4)
```

olelukoje, 2019-07-22:
@erivera1802, try this:
```
camera.CaptureRGBA(zeroCopy=1)
```

erivera1802, 2019-07-22:
> 
> 
> @erivera1802, try this:
> 
> ```
> camera.CaptureRGBA(zeroCopy=1)
> ```

Thanks! it worked!

jtiscione, 2020-02-21:
What does zeroCopy=1 mean?

dusty-nv, 2020-02-21:
It means to use zeroCopy memory (mapped CPU/GPU memory) to store the image, instead of GPU-only memory

HerrAskin, 2020-04-04:
Hi, all! I'm sorry, but I don't understand why I have to read an image from the cuda-memory (cudaToNumpy), cut out the area, and convert it back to the cuda (cudaFromNumpy)... Can't you just define the memory area to work with? The essence of the problem is that when the neural network is working, the entire frame is processed first, the object is detected, and then the neural network works only in this area... Converting to and from an array will take a very long time.

HerrAskin, 2020-04-04:
Dastin, say me please! In https://github.com/dusty-nv/jetson-utils/tree/master/cuda there are cudaCrop files. Is there any way I can use them for my task? I'm working on a jetson AGX Xavier. If this is possible, we would like a small example of using it in python.