dusty-nv/jetson-inference #357: /imagenet-camera: error while loading shared libraries: libnvcaffe_parser.so.3: cannot open shared object file: No such file or directory

yeshalot, 2019-07-12:
i all,
Formerly i was using my Jetson TX2 smoothly for both ./imagenet-camera and ./imagenet-console.
However, I had to re-flash my Jetson TX2 last week and try to run that command how ever a get the warning below,

/imagenet-camera: error while loading shared libraries: libnvcaffe_parser.so.3: cannot open shared object file: No such file or directory

i read some topics, i can not find the solution yet. i found smt like this in github:

"assuming that libre.so and librem.so are installed in /usr/local/lib,
can you please try this:

LD_LIBRARY_PATH=/usr/local/lib baresip" 

however there is no such lib file installed under directory, /usr/local/lib , What should i do?

Ubuntu 18.04// Cuda 10.0.166// Tensorflow 1.13.1// TensorRT 5.0.6// Python 3.6.8 are installed..

Thanx in advance,

dusty-nv, 2019-07-18:
Hi @yeshalot, which version of JetPack did you re-flash your Jetson TX2 with?

yeshalot, 2019-07-18:
hi @dusty-nv 

how can i check my JetPack Version rather then this way i dont know? i just found this code and can just get these info's it says L4T 32.1.0 [ JetPack UNKNOWN ]

nvidia@nvidia-desktop:~$ head -n 1 /etc/nv_tegra_release
# R32 (release), REVISION: 1.0, GCID: 14531094, BOARD: t186ref, EABI: aarch64, DATE: Wed Mar 13 07:41:08 UTC 2019

nvidia@nvidia-desktop:~/jetsonUtilities$ python jetsonInfo.py 
NVIDIA Jetson TX2
L4T 32.1.0 [ JetPack UNKNOWN ]
Board: t186ref
Ubuntu 18.04.2 LTS
Kernel Version: 4.9.140-tegra
CUDA 10.0.166

dusty-nv, 2019-07-18:
That appears to be JetPack 4.2 (L4T R32.1).

Can you see if you have the file `libnvcaffe_parser.so.3` under `/usr/lib/aarch64-linux/gnu` directory?

You might want to try running `sudo ldconfig` and then re-building the repo.

closed 2023-04-27 by dusty-nv