dusty-nv/jetson-inference #360: python wasn't found

Darcyzh, 2019-07-15:
I do as the [Building the Project from Source](https://github.com/dusty-nv/jetson-inference/blob/python/docs/building-repo-2.md) says, when it move on [Compiling the Project](https://github.com/dusty-nv/jetson-inference/blob/python/docs/building-repo-2.md#compiling-the-project), after I run 
`sudo make install`
 it shows that

>  -- trying to build python bindings for Python versions: 2.7;3.6;3.7 
>   -- detecting Python 2.7...
>   --Python 2.7 wasn't found
>   -- detecting Python 3.6...
>   --Python 3.6 wasn't found
>   -- detecting Python 3.7...
>   --Python 3.7 wasn't found

And this problem also occurred after run
`cmake ../` in [Configuring with CMake](https://github.com/dusty-nv/jetson-inference/blob/python/docs/building-repo-2.md#configuring-with-cmake)

After I finished the Building, I test as [Classifying Images with ImageNet](https://github.com/dusty-nv/jetson-inference/blob/python/docs/imagenet-console-2.md#classifying-images-with-imagenet) says. And it is OK with the command `./imagenet-console --network=googlenet orange_0.jpg output_0.jpg` that is a C++ program, but if I run `./imagenet-console.py --network=googlenet orange_0.jpg output_0.jpg`  that is a python program, the error arose:

> Traceback (most recent call last):
>     File "./imagenet-console.py", line 24, in <module>
>        import jetson.inference 
>   ImportError: No module named jetson.inference

More information: 

- In step [Downloading Models](https://github.com/dusty-nv/jetson-inference/blob/python/docs/building-repo-2.md#downloading-models), I didn't use the default Model Downloader tool because I'm in China, and I am unable to connect to Box.com.  So I download the neural network model in https://github.com/dusty-nv/jetson-inference/releases.
- My board is Jetson TX2, the onboard system is ubuntu 16.04, the Jetpack version is 3.3.
- The onboard ubuntu system has python 2.7 and 3.5. The default python is 2.7. If you type `python`, it is python 2.7.11+, if you type `python3`, it is python 3.5.2.

So what is the cause of it can't detecting Python and why the python version test program can't work, how can I fix the problem?
If anyone can help me, I'll appreciate it.

Darcyzh, 2019-07-16:
I've solved part of the problem. 
After I run:
`$ sudo apt-get install libpython-dev python-numpy`
python 2.7 can be found. It proved that the original onboard ubuntu system didn't comes with the `libpython-dev` and `python-numpy` packages pre-installed (which are for Python 2.7) as [Python Development Packages](https://github.com/dusty-nv/jetson-inference/blob/python/docs/building-repo-2.md#python-development-packages) described.
But after I install python3.6 and built soft link to the command `python`, with `libpython3-dev` and `python3-numpy` installed, the python 3.6 still can't be found. 
Anyone meet this problem?

Darcyzh, 2019-07-16:
@dusty-nv Could you give me a help?

Darcyzh, 2019-07-16:
I solved the problem that python 3.6 can't be found.
Just use:
`$ sudo apt-get install libpython3.6-dev`
instead of:
`$ sudo apt-get install libpython3-dev`
So the reason is the package version doesn't match.

dusty-nv, 2019-07-16:
Thanks Darcy - which version of JetPack and Ubuntu are you using?

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

Darcyzh, 2019-07-16:
Hello, dusty-nv.
  I use Ubuntu 16.04 LTS 64bit virtual host system within VMware and Jetpack 3.3 to flash OS onto Jetson TX2. The flashed OS on TX2 is also ubuntu 16.04 LTS.

------------------
Darcy

dusty-nv, 2019-07-16:
Ah I see, must be a difference in the package names for Ubuntu 16.04 on Jetson.

closed 2019-07-16 by Darcyzh