dusty-nv/jetson-inference #361: rgba to bgr conversion giving distorted image

Source82, 2019-07-15:
In an attempt to convert to bgr for further processing with opencv I wrote this code

cv::Mat im(imgHeight,imgWidth,CV_8FC4);
	cv::Mat converted;
	std::memcpy(im.data, imgCPU, imgWidth*imgHeight*sizeof(float));
	cv::cvtColor(im, converted, CV_RGB2BGR);
	cv::imwrite("result.jpg",im);
	cv::imwrite("result2.jpg",converted);

	
	std::cout << "Image Saved !" << std::endl;
Unfortunetly only the top part of the image is visible the rest appears black. Help please what am I doing wrong.

dusty-nv, 2019-07-15:
At a glance, I think you need to have `imgWidth*imgHeight*sizeof(float)*4` in your 3rd line of code.  Modified version is here: 

`std::memcpy(im.data, imgCPU, imgWidth*imgHeight*sizeof(float)*4);`

Source82, 2019-07-16:
Thanks that worked for me

closed 2019-07-16 by Source82

Source82, 2019-07-16:
even faster is replacing the memcpy with

im = cv::Mat(imgHeight, imgWidth,CV_32FC4, (void *)imgCPU);

komms, 2019-07-23:
@Source82 Thanks for the code.

 Even though I used the code, on a continous video stream, I somtimes get good images and sometimes not. Do you know, what could be the problem?

Source82, 2019-07-23:
What is the distortion like? can you let me see the code?

komms, 2019-07-24:
[image]

 cv::Mat im_CPU(camera->GetHeight(),camera->GetWidth(),CV_32FC4);
 cv::Mat converted_CPU;
 //std::memcpy(im_CPU.data, outCPU, camera->GetHeight()*camera->GetWidth()*sizeof(float)*4);
  im_CPU = cv::Mat(camera->GetHeight(), camera->GetWidth(), CV_32FC4, (void *)imgCPU);
  cv::cvtColor(im_CPU, converted_CPU, CV_RGBA2BGR);
  cv::imshow("converted", converted_CPU);
  cv::imwrite("RGBA_outcpu.jpg",im_CPU);
  cv::imwrite("RGB_outcpu.jpg",converted_CPU);
  std::cout << "outcpu Image Saved !" << std::endl;

Source82, 2019-07-24:
Am quite busy now, but I will try and run your code to see if I can re-create the error. By the way, I can see you are using imshow without waitkey, you are not giving highgui time to process the draw requests from cv::imshow().

aviontics, 2019-08-14:
I am having similar issue converting from imgRGBA to opencv mat.

Whould have thought there should be a standard function for this because it something a lot of people would want to do for further processing the detectNET camera example.

Please need help.

aviontics, 2019-08-14:
> I am having similar issue converting from imgRGBA to opencv mat.
> 
> Whould have thought there should be a standard function for this because it something a lot of people would want to do for further processing the detectNET camera example.
> 
> Please need help.

Solved the problem.
this worked for me.

                 float* imgRGBA = NULL;
		void* imgCPU = NULL;
		void* imgCUDA = NULL;
		float* imgRGBAX = NULL;
		if( !camera->Capture(&imgCPU, &imgCUDA, 1000) )
			printf("\ndetectnet-camera:  failed to capture frame\n");
		
		if( !camera->ConvertRGBA(imgCUDA, &imgRGBA) ){
				printf("detectnet-camera:  failed to convert from NV12 to RGBA\n");
		}

cv::Mat im = cv::Mat(imgHeight, imgWidth,CV_8UC3, (void *)imgCPU);