dusty-nv/jetson-inference #362: Multi-class Object Detection Models trained with DIGITS??

vxgu86, 2019-07-16:
how to do Multi-class Object Detection Models??
asking this question because I found training Multi-class Object Detection Models so difficult(maybe sth was wrong with my config) with DIGITS.
did I missing something, cause I could not even train a two-objects detection model effectively with DIGITS.

agoila, 2019-08-14:
Please refer to similar posts [here](https://github.com/NVIDIA-AI-IOT/jetson-trashformers/wiki/Single-and-Multi-Class-Object-Detection) and [here](https://lesterlo.wordpress.com/2017/05/10/train-multiple-classes-in-detectnet/).

vxgu86, 2019-08-30:
> Please refer to similar posts [here](https://github.com/NVIDIA-AI-IOT/jetson-trashformers/wiki/Single-and-Multi-Class-Object-Detection) and [here](https://lesterlo.wordpress.com/2017/05/10/train-multiple-classes-in-detectnet/).

does this make sense for the segmentation model？

closed 2023-04-27 by dusty-nv