dusty-nv/jetson-inference #365: ./download-models.sh:command not found

zoulu995, 2019-07-17:
When i want to use Model Downloader tool or install pytoch, i need to run follow command in Ubuntu terminal, like 
"
$ cd jetson-inference/tools
$ ./download-models.sh
"
or
"
$ cd jetson-inference/build
$ ./install-pytorch.sh
"
but it not work, what should i do now?
thanks for your help

dusty-nv, 2019-07-18:
What errors do you receive when you try to run the command?

zoulu995, 2019-07-19:
> Hi @zoulu995, what errors do you receive when you try to run the command?
Thanks for your replying.

At first I met this problem. when i enter command `$ ./download-models.sh`
it return
`-bash: ./download-models.sh: /bin/bash^M: bad interpreter: No such file or directory`

Then I found the following solutions Online
**The reason is that the format of the file is dos, so it's OK to change it to unix.
View the file format, open the file with vim, press ESC key, press Shift + ：, enter "set ff" and return. You can see the file format file format = DOS
Enter "set ff = UNIX" by shift + ： and find no response. That's right.
Fileformat = UNIX can be viewed by shift + colon set FF**

But after that，I enter the command `$ ./download-models.sh`
it return 

```
znqc@znqc-desktop:~/PKU/Zouyuhang/github/jetson-inference/tools$ ./download-models.sh 
[jetson-inference]  Checking for 'dialog' deb package...not installed
[jetson-inference]  FOUND_DIALOG=
./download-models.sh: line 553: dialog: command not found
[jetson-inference]  Model selection status:  127
[jetson-inference]  Model selection cancelled.
 
[jetson-inference]  to run this tool again, use the following commands:
 
    $ cd <jetson-inference>/tools
    $ ./download-models.sh

```
And then no matter how I enter this command ` $ ./download-models.sh` There are still problems like this.

 what should i do now .

dusty-nv, 2019-07-23:
Hi @zoulu995, try running `sudo apt-get install dialog`, and then run the script again.

This should be installed by the script for you, but seems like it isn't for some reason.

IeiuniumLux, 2020-03-26:
@dusty-nv  `sudo apt-get install dialog` fixed for me. Now when I run the script, the dialog shows up. Thank you!
[image]

closed 2023-04-27 by dusty-nv