dusty-nv/jetson-inference #366: cuda memory in multiprocessing.Queue: TypeError: can't pickle PyCapsule objects

benkelaci, 2019-07-17:
If I try to put to a cuda memory frame to a multiprocessing.Queue() (self.q_numpy_frame)  in a separate process, then I get "TypeError: can't pickle PyCapsule objects"
```
    color_image = cv2.cvtColor(frame, cv2.COLOR_BGR2RGBA)
    cuda_frame = jetson_utils_python.cudaFromNumpy(color_image)
    self.q_numpy_frame.put((cuda_frame, width, height))
```
The trackeback:
```
Traceback (most recent call last):
  File "/usr/lib/python3.6/multiprocessing/queues.py", line 234, in _feed
    obj = _ForkingPickler.dumps(obj)
  File "/usr/lib/python3.6/multiprocessing/reduction.py", line 51, in dumps
    cls(buf, protocol).dump(obj)
TypeError: can't pickle PyCapsule objects
```
Do you have a clue why? What is the workaround? Thanks.

dusty-nv, 2019-07-18:
Hi @benkelaci , it would seem that the `multiprocessing` package is incapable of handling the PyCapsule containers that store the CUDA memory pointer.  In lieu of that, you would probably want to pass it an numpy ndarray instead.  You could call `jetson.utils.cudaFromNumpy()` on that ndarray at a later time if/when you wanted to pass it to `jetson.inference` package for inferencing.

benkelaci, 2019-07-18:
Yeah, but my assumption goal is one process making the cudaFromNumpy(), meanwhile another process can make the inference of the previous frame. Like this I might get better (faster) end-to-end running time (conversion+inference). Can't I pass it somehow or reduce the cudaFromNumpy() time (it is aroud 20-30ms)?

dusty-nv, 2019-07-18:
It doesn't appear that `multiprocessing` supports the PyCapsule objects - could you try using the Python `thread` instead to manually spawn threads to pass around the object returned from `cudaFromNumpy()`?

closed 2023-04-27 by dusty-nv