dusty-nv/jetson-inference #367: Problem with CaptureRGBA() frame to numpy convertation

olelukoje, 2019-07-19:
After capturing:
`frame, width, height = camera.CaptureRGBA()`
I want to get a numpy array from the frame:
`arr = jetson.utils.cudaToNumpy(frame, width, height, 4)`
But get this exception:
`Traceback (most recent call last):
  File "detectnet-camera-recognition.py", line 91, in <module>
    frame = jetson.utils.cudaToNumpy(frame, width, height, 4)
Exception: jetson.utils -- cudaToNumpy() failed to get input array pointer from PyCapsule container
PyTensorNet_Dealloc()
jetson.utils -- PyCamera_Dealloc()`

Is there a way to convert a frame to numpy? Thank you.

bwsw, 2019-07-22:
Use 
```
camera.CaptureRGBA(zeroCopy=1)
```

It works.

closed 2019-07-22 by olelukoje

jtiscione, 2020-02-21:
Where is the documentation for this method? All I've been able to find is https://rawgit.com/dusty-nv/jetson-inference/python/docs/html/python/jetson.utils.html which is just a list of method names.