dusty-nv/jetson-inference #368: Jetson TX2 Library Path not set/updated

rdejana, 2019-07-19:
After building, I can see the so's copied over to /usr/local/lib/, however that directory is not in the library path.  

```
/data/jetson-inference/build/aarch64/bin$ ./imagenet-console.py granny_smith_1.jpg output_1.jpg
jetson.inference.__init__.py
Traceback (most recent call last):
  File "./imagenet-console.py", line 24, in <module>
    import jetson.inference
  File "/usr/lib/python2.7/dist-packages/jetson/inference/__init__.py", line 4, in <module>
    from jetson_inference_python import *
ImportError: libjetson-utils.so: cannot open shared object file: No such file or directory
```
Once the lib path is set, example runs fine.

dusty-nv, 2019-07-23:
If you were to run `sudo ldconfig`, that may also take care of it - although this didn't seem to be necessary for me using recent JetPack.  Just curious, which version of JetPack are you using?

rdejana, 2019-07-25:
that looks good, closing out.  Thanks Dusty!

closed 2019-07-25 by rdejana