dusty-nv/jetson-inference #371: Does this package support Caffe model only?

deeprun, 2019-07-25:
Does this package support tensorflow?

dusty-nv, 2019-07-26:
It supports Caffe, ONNX, and TensorFlow UFF

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

ArtlyStyles, 2019-07-26:
Is this considered TensorFlow UFF? https://github.com/MaybeShewill-CV/lanenet-lane-detection

ArtlyStyles, 2019-07-29:
@dusty-nv could you please let me know if tensorflow ckpt is UFF and can be used for this package? Or how can I convert ckpt to UFF?

closed 2023-04-27 by dusty-nv