dusty-nv/jetson-inference #374: cat-dog-dataset.sh in tools directory won't work as expected

Jackson-Rong, 2019-07-27:
Hello! I run the script, and it keep telling me:
  1. ./cat-dog-dataset.sh: line 34: /usr/bin/shuf: Argument list too long
cp: missing file operand
 2. rename function not working : change JPEG to jpg.  I change the 'rename' command in the script to 'rename.ul', but it still won't work.

Has this script been throughly tested ?

dusty-nv, 2019-07-27:
It was developed and working on my Ubuntu 16.04 host system.  Are you using it on 18.04 or another OS?

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

Jackson-Rong, 2019-07-27:
Hello!
  Thanks for the quick response.  I was running  the script on Jetson Nano development kit, Jetpack 4.2.1, believed to be Ubuntu 18.04.

dusty-nv, 2019-07-27:
Unfortunately I am away from my systems and can't test it at this time.  I do recall including a link in the source to where I found the command for randomly selecting the files.  If unable to debug, you might find it easier to create a new one or pull the files manually.

Jackson-Rong, 2019-07-27:
Hello! Dustin,
   Thanks.  I do find another interesting thing when running the next script:checking every plants in the test directory. The script failed and I got the message: cd: ~/datasets/PlantCLEF_Subset: No such file or directory  cp: cannot stat 'test': No such file or directory
  It seems the bash shell can not recognize the '~' character. Maybe you are right, Jetson Nano Jetpack was a minimized Ubuntu 18.04, some function might be accidentically been removed.
Best Regards,Jackson Rong

SimoGira, 2019-07-31:
Hello! @Jackson-Rong 
Try to replace the '~' symbols with the $HOME variable (on Ubuntu it points to the home directory by default)
Then I found a possible error in labels path, where I need to remove the '../' in labels=$DATASET/../labels.txt $f $f

This is the new script:
`#!/bin/bash  
NET="$HOME/jetson-inference/python/training/imagenet/plants_epoch_100"
DATASET="$HOME/datasets/PlantCLEF_Subset"

cd $DATASET
cp -r test test_output

FILES="$DATASET/test_output/*.jpg"

cd $DATA

for f in $FILES
do
     echo "Processing $f"
     imagenet-console --model=$NET/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=$DATASET/labels.txt $f $f
done`

Jackson-Rong, 2019-08-01:
Hello! @SimoGira ,

   Thanks.  The new script works well.  Now only the cat-dog-dataset.sh still won't work.   It seems the redirection command caused some problem in cp function.

closed 2023-04-27 by dusty-nv