dusty-nv/jetson-inference #375: error "cannot find -lGL" during building jetson-inference on nano board

flycat0101, 2019-07-28:
hello, 
I used JetPack4.2.1 on Jetson Nano board, when I tried to build the jetson-inference repos on nano board, I got below error, how to resolve it?

/usr/bin/ld: cannot find -lGL
collect2: error: ld returned 1 exit status
utils/CMakeFiles/jetson-utils.dir/build.make:2627: recipe for target 'aarch64/lib/libjetson-utils.so' failed
make[2]: *** [aarch64/lib/libjetson-utils.so] Error 1
CMakeFiles/Makefile2:927: recipe for target 'utils/CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [utils/CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

thimabru1010, 2019-11-29:
I found the same error

Arcadia-1, 2020-01-08:
I found the same error

SaddamBInSyed, 2020-01-27:
@dusty-nv could you please advise about the below error.?
/usr/bin/ld: cannot find -lGL
collect2: error: ld returned 1 exit status
utils/CMakeFiles/jetson-utils.dir/build.make:2627: recipe for target 'aarch64/lib/libjetson-utils.so' failed
make[2]: *** [aarch64/lib/libjetson-utils.so] Error 1
CMakeFiles/Makefile2:927: recipe for target 'utils/CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [utils/CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

huanglidongTX2, 2020-02-13:
so,have anyone solved this problem? I found the same error

flurpo, 2020-06-14:
I have the same error running make on my TX2, can install inference...../jetson-inference/build/make

root@tx2:/jetson-inference/build# make
[  1%] Linking CXX shared library ../aarch64/lib/libjetson-utils.so
/usr/bin/ld: cannot find -lGL
collect2: error: ld returned 1 exit status
utils/CMakeFiles/jetson-utils.dir/build.make:2627: recipe for target 'aarch64/lib/libjetson-utils.so' failed
make[2]: *** [aarch64/lib/libjetson-utils.so] Error 1
CMakeFiles/Makefile2:927: recipe for target 'utils/CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [utils/CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

flurpo, 2020-06-14:
I'm able to see libGL.so.

@SaddamBInSyed tx2:/jetson-inference$ locate libGL.so
/usr/lib/aarch64-linux-gnu/libGL.so
/usr/lib/aarch64-linux-gnu/libGL.so.1
/usr/lib/aarch64-linux-gnu/libGL.so.1.0.0

I'm also able to see the component path aarch64/lib/libjetson-utils.so, the target is there.
?
looks like a problem with 'all'?

dusty-nv, 2020-06-14:
Hi Mark, can you check this post?

https://forums.developer.nvidia.com/t/no-usr-lib-aarch64-linux-gnu-tegra-libgl-so-0-with-jetpack-4-4-1/67382/6?u=dusty_nv

Sometimes installing other libraries will overwrite libGL.  You can also check 'ls -ll' to see if the symlink is broken.

I believe libglvnd is in apt, you can try re-installing it.

flurpo, 2020-06-16:
Dusty-nv,

Thanks so much for such a quick response and the link.

-I'm 18.04 so no /usr/local/lib/aarch64-linux-gnu/libGL.so?

-/usr/lib/aarch64-linux-gnu/
ls -ll libGL*

-lrwxrwxrwx 1 root root      14 Jun 11 21:50 libGL.so -> tegra/libGL.so
lrwxrwxrwx 1 root root      14 May 10  2019 libGL.so.1 -> libGL.so.1.0.0
-rw-r--r-- 1 root root  972968 May 10  2019 libGL.so.1.0.0

-Tried to apt libglvnd.
m@tx2:/$ sudo apt-get install libglvnd
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libglvnd

Let me know your thoughts. Best,

flurpo, 2020-06-16:
Dusty-nv,

It appears that the link is not just broken....libGL.so is not in the path /tegra/libGL.so!

markw@tx2:/usr/lib/aarch64-linux-gnu$ ls -lL libGL.*
ls: cannot access 'libGL.so': No such file or directory
-rw-r--r-- 1 root root 972968 May 10  2019 libGL.so.1
-rw-r--r-- 1 root root 972968 May 10  2019 libGL.so.1.0.0

Is the libGL.so file that is supposed to be contained in /tegra customized for jetson-inference?

I've seen other OpenCV installations (i.e. OpenCVTX2 Github) where they recommend erasing this symlink and creating a new one to libGL.so.1. ( a link to a link?)

Thanks for your insight. Let me know what you think. Best.

dusty-nv, 2020-06-16:
Hi Mark, here is what mine looks like:

```bash
lrwxrwxrwx 1 root root      14 May 10  2019 libGL.so -> libGL.so.1.0.0
lrwxrwxrwx 1 root root      14 May 10  2019 libGL.so.1 -> libGL.so.1.0.0
-rw-r--r-- 1 root root  972968 May 10  2019 libGL.so.1.0.0
```

Try `sudo ln -s /usr/lib/aarch64-linux-gnu/libGL.so.1.0.0 /usr/lib/aarch64-linux-gnu/libGL.so`

flurpo, 2020-06-16:
Dusty-nv,

sudo rm /usr/lib/aarch64-linux-gnu/libGL.so
sudo ln -s /usr/lib/aarch64-linux-gnu/libGL.so.1.0.0 /usr/lib/aarch64-linux-gnu/libGL.so

Thanks so much. That was it! 
Linking successful!

For anyone monitoring I tried to install OpenCVTx2.sh which pointed my symlink libGL.so to /tegra. Be Careful with that item.

Thanks to Dusty-nv all corrected. Best.

closed 2023-04-27 by dusty-nv