dusty-nv/jetson-inference #378: How do I implement own model of transfer learning?

duttasantanuGH, 2019-08-04:
Hi Dusty

Can you please provide me with a alternate resource which can help me use and adopt train.py under jetson-inference... to implement own model for transfer learning? Basically I would like to experiment to see if we freeze most of layers of pre-trained model, and add some extra layer on top of pre-trained layer and then train network with own images - does it improve accuracy?

Thanks in advance for your kind help in anticipation.

Regards
Santanu

dusty-nv, 2019-08-04:
Hi Santanu, you might find this PyTorch tutorial on Transfer Learning useful:

https://pytorch.org/tutorials/beginner/transfer_learning_tutorial.html

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

duttasantanuGH, 2019-08-07:
Hi Dusty
Thanks for your advice. I have used another blog on Sports-Type-Classifier and already have a keras trained model. I have converted keras model to onnx using keras2onnx. But when I am trying to use the model, I am getting WARNING: ONNX model has a newer ir_version (0.0.5) than this parser was built against (0.0.3) followed by the below error:
Input filename:   Sports-Type-Classifier/modelsd1.onnx
ONNX IR version:  0.0.5
Opset version:    7
Producer name:    keras2onnx
Producer version: 1.5.1
Domain:           onnx
Model version:    0
Doc string:       
----------------------------------------------------------------
WARNING: ONNX model has a newer ir_version (0.0.5) than this parser was built against (0.0.3).
While parsing node number 5 [Add -> "biased_tensor_name1"]:
ERROR: /home/erisuser/p4sw/sw/gpgpu/MachineLearning/DIT/release/5.0/parsers/onnxOpenSource/builtin_op_importers.cpp:332 In function importAdd:
[8] Assertion failed: attrs.count("broadcast")
[TRT]   failed to parse ONNX model 'Sports-Type-Classifier/modelsd1.onnx'
[TRT]   device GPU, failed to load Sports-Type-Classifier/modelsd1.onnx
[TRT]   failed to load Sports-Type-Classifier/modelsd1.onnx
[TRT]   imageNet -- failed to initialize.
jetson.inference -- imageNet failed to load built-in network 'googlenet'
PyTensorNet_Dealloc()
Traceback (most recent call last):
  File "/usr/local/bin/imagenet-console.py", line 53, in <module>
    net = jetson.inference.imageNet(opt.network, argv)
Exception: jetson.inference -- imageNet failed to load network
jetson.utils -- freeing CUDA mapped memory
Can you please help me in resolving this?
Thanks and Regards
Santanu

duttasantanuGH, 2019-08-11:
Hi Dusty
With some effort, I implemented pytorch model outside and then converted pytorch model using jetson-inference helper function. But when I am doing inference confidence value is coming to be *100:
imageNet -- loaded 3 class info entries
Sports-Type-Classifier/s10_model.onnx initialized.
class 0002 - 2.217339  (formula1)
image is recognized as 'formula1' (class #2) with 221.733904% confidence

Can you please resolve it? Thanks in advance.
Regards
Santanu

closed 2023-04-27 by dusty-nv