dusty-nv/jetson-inference #381: Run another model outside this project in parallel

yeyicheng, 2019-08-05:
Hi, I am using detectNet-camera.py to detect objects. And I have another project recognizing faces in the frame using TensorFlow. Is it possible to run the two models in parallel?

Thank you!

dusty-nv, 2019-08-05:
Yes, you could launch another thread to run your other model in parallel.  Or run them in a pipeline serially.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

yeyicheng, 2019-08-05:
@dusty-nv Could you show me some code example please? I am new to this. Thank you.

dusty-nv, 2019-08-05:
I dont have one of that specific scenario with TF.  What I would start with, is just call the jetson.inference model and your TensorFlow model from the same script (same thread). Both utilize the GPU and likely will get serialized in the end anyways (although if there is CPU processing in TensorFlow model, that could benefit from running in a separate thread).  So you could start with them both in a single thread (same script).

yeyicheng, 2019-08-05:
@dusty-nv I tried to add the TensorFlow detection code right before `display.RenderOnce(img, width, height)`, but it does not seem to work. The system freezed and I have to restart.
```
while display.IsOpen():
	# capture the image
	img, width, height = camera.CaptureRGBA()

	# detect objects in the image (with overlay)
	detections = net.Detect(img, width, height)

	# print the detections
	print("detected {:d} objects in image".format(len(detections)))

	for detection in detections:
		print(detection)

	# render the image
	display.RenderOnce(img, width, height)

	# update the title bar
	display.SetTitle("{:s} | Network {:.0f} FPS".format(opt.network, 1000.0 / net.GetNetworkTime()))

	# synchronize with the GPU
	if len(detections) > 0:
		jetson.utils.cudaDeviceSynchronize()

	# print out performance info
	net.PrintProfilerTimes()
```

yeyicheng, 2019-08-06:
Actually I want to detect objects and faces in one window. Anyone can give an example?

closed 2023-04-27 by dusty-nv