dusty-nv/jetson-inference #383: imagenet-camera.py - saving frame delivers error

SlickJackson, 2019-08-11:
Hey there,
I'm already several hours working on the problem but can't fix it.
...I tried to save a picture if a specific class was detected:
`
#!/usr/bin/python
import jetson.inference
import jetson.utils
import argparse
...
...`

everything the same until the end

`

        # classify the image
        class_idx, confidence = net.Classify(img, width, height)

        # find the object description
        class_desc = net.GetClassDesc(class_idx)

        # overlay the result on the image       
        font.OverlayText(img, width, height, "{:05.2f}% {:s}".format(confidence * 100, class_desc), $

        # render the image
        display.RenderOnce(img, width, height)

        # update the title bar
        display.SetTitle("{:s} | Network {:.0f} FPS".format(net.GetNetworkName(), 1000.0 / net.GetNe$

        # print out performance info
        net.PrintProfilerTimes()

        # write image to jpeg file if class is visible
        if class_idx == 673:
                jetson.utils.saveImageRGBA("test.png", img, width, height)
                print("picture was saved")
        else:
                print("nothing was saved")

...so I just added the last part. 
Index 673 is a computer mouse, just for testing.
Everytime I run the code, it is normally working until the mouse is detected, after that I get the following error:
`Traceback (most recent call last):
  File "./recognition.py", line 74, in <module>
    jetson.utils.saveImageRGBA("test.png", img, width, height)
Exception: jetson.utils -- saveImageRGBA() failed to get input image pointer from PyCapsule container`

...I don't know what I'm doing wrong.
I want to collect pictures for special events to use em later on for further training.

Thanks a lot!

dusty-nv, 2019-08-11:
Hi SlickJackson, try passing the argument zeroCopy=1 to camera.CaptureRGBA()

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

SlickJackson, 2019-08-11:
Thanks a lot dusty-nv!... u saved my day :)...well, evening in germany,
case closed!

kenken64, 2019-12-04:
zeroCopy=1  saving black and white square box any idea why?

dusty-nv, 2019-12-04:
> zeroCopy=1 saving black and white square box any idea why?

Hi @kenken64, can you post your code that you are using?

You might want to try a call to `jetson.utils.cudaDeviceSynchronize()` after `camera.CaptureRGBA()` and before you save the image.

closed 2023-04-27 by dusty-nv